Ysurac/openmptcprouter #1: little help and finish the documentation

olaulau, 2018-02-20:
Hi ;

To start, thank for the great job. Installation is very easy and is seems very complete (and similar to something ...)
I successfully complete VPS install (just had to change eth0 -> enp2s0 in 2 shorewall config files before rebooting).
On the routeur side, I'm using .vdi image in virtualbox, following the doc : WAN1 & WAN2 conf, Shadowsocks conf.

I think the doc lacks the GloryTun conf.
Maybe you should add some steps to test too.

While configured, my glorytun won't start.
On the summary page, it says that ipv4 and ipv6 network are not connected.
I can't "ping 8.8.8.8" from the router term.

On the interface page, status column isn't able to fetch datas.
There are some javascript syntax error in the firefox console.

Ysurac, 2018-02-20:
Hi,
I will try to find the default interface for VPS script.

I added glorytun conf 30 minutes ago :)

If you can't ping from the router than it's not working. Did you enable ss-redir ? (I added this step in the doc).
Make sure you can ping all IPs router from the router.

On the interface page I think you are using french translation, I need to find how to fix this.

olaulau, 2018-02-20:
thanks for the doc update.

got a firefox plugin that forces local to en-us, now interface status column is working.
is it so obvious I'm french ?

glorytun is now properly starting.
but in shawdowsocks redir rules page, dropdown seems to be wrong, here is what I have :
- hi2 - tcp_and_udp
- hi3 - tcp_and_udp
- hi4 - tcp_and_udp
- <unset>

still can't ping internet from the router term, neither from a computer configured to use the router.

Ysurac, 2018-02-20:
French translation will be fixed in next release. I'm french too and it's the only translation available for now.

I will update screenshot for shadowsocks rules asap.

There is a bug with the failover script that put the correct default route. If gateway is on a down interface, the script do nothing... This will be fixed in next release.

Next release will be available when compiled by CircleCI so in about 3H...

olaulau, 2018-02-20:
I've just tried with 0.6 .vdi image, still doesn't work :-/

Ysurac, 2018-02-21:
Both wan are up ?
Did you try a reboot ?
What do you have in system log ?

olaulau, 2018-02-21:
just retried with router .vdi v0.6.2, still doesn't work.
Is the VPS install script updated ? do I have to reinstall if too ?

in system / overview :
```
IPv4 WAN Status | ? 				Not connected | ? | Not connected
-- | -- | -- | --
? | Not connected
IPv6 WAN Status | ? 				Not connected | ? | Not connected
? | Not connected
Active Connections | 46 / 131072 (0%)
```

end of syslog :
```
Wed Feb 21 08:21:20 2018 daemon.warn dnsmasq[3085]: no servers found in /tmp/resolv.conf.auto, will retry
Wed Feb 21 08:21:20 2018 daemon.info dnsmasq[3085]: read /etc/hosts - 4 addresses
Wed Feb 21 08:21:20 2018 daemon.info dnsmasq[3085]: read /tmp/hosts/odhcpd - 0 addresses
Wed Feb 21 08:21:20 2018 daemon.info dnsmasq[3085]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Wed Feb 21 08:21:20 2018 daemon.info dnsmasq-dhcp[3085]: read /etc/ethers - 0 addresses
Wed Feb 21 08:21:20 2018 user.notice unbound: iterator will use built-in root hints
Wed Feb 21 08:21:20 2018 daemon.notice unbound: [3168:0] notice: init module 0: iterator
Wed Feb 21 08:21:21 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:22 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:23 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:23 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.10.1 dev wan1
Wed Feb 21 08:21:24 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:26 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:26 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.11.1 dev wan2
Wed Feb 21 08:21:30 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:32 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:32 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.10.1 dev wan1
Wed Feb 21 08:21:35 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:37 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:37 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.11.1 dev wan2
Wed Feb 21 08:21:40 2018 daemon.info procd: Instance mptcp::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Wed Feb 21 08:21:41 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:43 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:43 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.10.1 dev wan1
Wed Feb 21 08:21:44 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:46 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:46 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.11.1 dev wan2
Wed Feb 21 08:21:49 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:50 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:51 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:52 2018 daemon.err omr-tracker[2225]: ping: bad address ''
Wed Feb 21 08:21:52 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.10.1 dev wan1
Wed Feb 21 08:21:53 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:55 2018 daemon.err omr-tracker[2224]: ping: bad address ''
Wed Feb 21 08:21:55 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.11.1 dev wan2
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1720]: listening at 0.0.0.0:1100
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1718]: listening at 0.0.0.0:1100
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1720]: tcp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1718]: tcp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1718]: UDP relay enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1720]: UDP relay enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1720]: udp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1718]: udp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1720]: running from root user
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1718]: running from root user
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1719]: listening at 0.0.0.0:1100
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1719]: tcp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1719]: UDP relay enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1719]: udp port reuse enabled
Wed Feb 21 08:21:58 2018 daemon.info /usr/bin/ss-redir[1719]: running from root user
Wed Feb 21 08:21:58 2018 daemon.info glorytun[2165]: INITIALIZED tun0
Wed Feb 21 08:21:58 2018 daemon.notice netifd: Interface 'glorytun' is enabled
Wed Feb 21 08:21:58 2018 daemon.notice netifd: Network device 'tun0' link is up
Wed Feb 21 08:21:58 2018 daemon.notice netifd: Interface 'glorytun' has link connectivity 
Wed Feb 21 08:21:58 2018 daemon.notice netifd: Interface 'glorytun' is setting up now
Wed Feb 21 08:21:58 2018 daemon.notice netifd: Interface 'glorytun' is now up
Wed Feb 21 08:21:58 2018 kern.notice kernel: [   49.430600] random: nonblocking pool is initialized
Wed Feb 21 08:21:58 2018 daemon.info unbound: [3168:0] info: start of service (unbound 1.6.8).
[... cut ...]
[    4.491124] kmodloader: done loading kernel modules from /etc/modules.d/*
[    5.627554] ip_local_port_range: prefer different parity for start/end values.
[    6.181406] IPv6: ADDRCONF(NETDEV_UP): lo: link is not ready
[    6.182690] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.185964] 8021q: adding VLAN 0 to HW filter on device eth0
[    6.188865] 8021q: adding VLAN 0 to HW filter on device wan1
[    6.190411] 8021q: adding VLAN 0 to HW filter on device wan2
[   49.430600] random: nonblocking pool is initialized
```

Ysurac, 2018-02-21:
VPS script not updated.
The state in system overview is not a problem.
Multipath should be set as disabled for glorytun interface. (same for lan interface)
You should check that shadowsocks key are the same on VPS and OpenMPTCProuter: You can check on the OpenMPTCProuter ```cat /tmp/etc/shadowsocks-libev/ss_redir.hi.json``` and ```cat /etc/shadowsocks-libev/config.json``` on the VPS, key should be the same.

olaulau, 2018-02-21:
shadowsocks key are identical.

disabled multipath on wan1 & wan2
 -> now ping says "network unreachable" instead of doing nothing.

But I still don't have exactly "hi" in redir rules, but hi2, ui3, hi4, and unset in the dropdown.
I had to enable it in "local instance" tab of shadowsocks, then select "hi" in redir rules
-> no change.

olaulau, 2018-02-21:
oh sorry for wan* mistake ... re-enabled multipath on them.
multipath was disabled for lan and glorytun.

content of /etc/config/shadowsocks-libev :
```
config ss_redir 'hi'
	option server 'sss0'
	option local_address '0.0.0.0'
	option local_port '1100'
	option mode 'tcp_and_udp'
	option timeout '60'
	option fast_open '1'
	option verbose '1'
	option reuse_port '1'
	option mptcp '1'
	option disabled 'false'

config ss_rules 'ss_rules'
	option src_default 'forward'
	option dst_default 'forward'
	option local_default 'forward'
	list dst_ips_forward '8.8.8.8'
	option redir_tcp 'hi'
	option redir_udp 'hi'

config server 'sss0'
	option server_port '65101'
	option method 'aes-256-cfb'
	option server 'my_vps_ip'
	option key 'my_shadowsocks_key'

config ss_tunnel 'dns'
	option disabled '1'
	option mode 'tcp_and_udp'
	option server 'sss0'
	option local_port '5353'
	option tunnel_address '8.8.8.8:53'

config ss_redir 'hi2'
	option server 'sss0'
	option local_address '0.0.0.0'
	option local_port '1100'
	option mode 'tcp_and_udp'
	option timeout '60'
	option fast_open '1'
	option reuse_port '1'
	option mptcp '1'

config ss_redir 'hi3'
	option server 'sss0'
	option local_address '0.0.0.0'
	option local_port '1100'
	option mode 'tcp_and_udp'
	option timeout '60'
	option fast_open '1'
	option reuse_port '1'
	option mptcp '1'

config ss_redir 'hi4'
	option server 'sss0'
	option local_address '0.0.0.0'
	option local_port '1100'
	option mode 'tcp_and_udp'
	option timeout '60'
	option fast_open '1'
	option reuse_port '1'
	option mptcp '1'
```

Ysurac, 2018-02-21:
This configuration is ok.

Maybe a problem on the VPS part ? Shorewall (the firewall part) is running on the VPS ?
eth0 is replaced by enp2s0 in all shorewall conf files (interfaces and snat) ?

olaulau, 2018-02-21:
service shorewall status
```
● shorewall.service - Shorewall IPv4 firewall
   Loaded: loaded (/lib/systemd/system/shorewall.service; enabled; vendor preset
   Active: active (exited) since Wed 2018-02-21 13:39:37 CET; 10min ago
  Process: 1747 ExecStop=/sbin/shorewall $OPTIONS clear (code=exited, status=0/S
  Process: 1855 ExecStart=/sbin/shorewall $OPTIONS start $STARTOPTIONS (code=exi
 Main PID: 1855 (code=exited, status=0/SUCCESS)

Feb 21 13:39:37 new2 shorewall[1855]: Starting Shorewall....
Feb 21 13:39:37 new2 shorewall[1855]: Initializing...
Feb 21 13:39:37 new2 shorewall[1855]: Setting up Route Filtering...
Feb 21 13:39:37 new2 shorewall[1855]: Setting up Martian Logging...
Feb 21 13:39:37 new2 shorewall[1855]: Setting up Accept Source Routing...
Feb 21 13:39:37 new2 shorewall[1855]: Preparing iptables-restore input...
Feb 21 13:39:37 new2 shorewall[1855]: Running /sbin/iptables-restore ...
Feb 21 13:39:37 new2 shorewall[1855]: IPv4 Forwarding Enabled
Feb 21 13:39:37 new2 shorewall[1855]: done.
Feb 21 13:39:37 new2 systemd[1]: Started Shorewall IPv4 firewall.
```
`iptables -L` shows many rules, so I think shorewall is working correctly.

on SSH login I have : 
`< OpenMPCTProuter VPS 0.2 >`

I'm using a fresh debian9 install on an online.net dedicated server.
Just ran the VPS script, edit shorewall interface, and reboot.
Then I use the keys generated into the router config (shadowsocks & glorytun).

still cannot ping anything : 
```
root@OpenMPTCProuter:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
ping: sendto: Network unreachable
```
any idea ?

Ysurac, 2018-02-21:
Did you edit also shorewall snat ?

Ysurac, 2018-02-21:
What is the result if you stop shorewall ?

olaulau, 2018-02-21:
/etc/shorewall/interfaces : 
```
###############################################################################
?FORMAT 2
###############################################################################
#ZONE   INTERFACE       OPTIONS
net	enp2s0          dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn	gt-tun0        nosmurfs,routefilter,logmartians,tcpflags
```
/etc/shorewall/snat : 
```
###############################################################################
?FORMAT 2
###############################################################################
#ZONE   INTERFACE       OPTIONS
net	enp2s0          dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn	gt-tun0        nosmurfs,routefilter,logmartians,tcpflags
```

stopping shorewall : 
```
# service shorewall stop
# iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
```
still cannot ping internet from router

restarting shorewall refills iptables with many rules : 
```
root@new2:~# iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination         
net-fw     all  --  anywhere             anywhere            
vpn-fw     all  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere            
Reject     all  --  anywhere             anywhere            
LOG        all  --  anywhere             anywhere             LOG level info prefix "Shorewall:INPUT:REJECT:"
reject     all  --  anywhere             anywhere            [goto] 

Chain FORWARD (policy DROP)
target     prot opt source               destination         
net_frwd   all  --  anywhere             anywhere            
vpn_frwd   all  --  anywhere             anywhere            
Reject     all  --  anywhere             anywhere            
LOG        all  --  anywhere             anywhere             LOG level info prefix "Shorewall:FORWARD:REJECT:"
reject     all  --  anywhere             anywhere            [goto] 

Chain OUTPUT (policy DROP)
target     prot opt source               destination         
fw-net     all  --  anywhere             anywhere            
fw-vpn     all  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere            
Reject     all  --  anywhere             anywhere            
LOG        all  --  anywhere             anywhere             LOG level info prefix "Shorewall:OUTPUT:REJECT:"
reject     all  --  anywhere             anywhere            [goto] 

Chain Broadcast (2 references)
target     prot opt source               destination         
DROP       all  --  anywhere             anywhere             ADDRTYPE match dst-type BROADCAST
DROP       all  --  anywhere             anywhere             ADDRTYPE match dst-type MULTICAST
DROP       all  --  anywhere             anywhere             ADDRTYPE match dst-type ANYCAST

Chain Drop (2 references)
[... 131 lines cut ...]

Chain vpn-fw (1 references)
target     prot opt source               destination         
dynamic    all  --  anywhere             anywhere             ctstate INVALID,NEW,UNTRACKED
smurfs     all  --  anywhere             anywhere             ctstate INVALID,NEW,UNTRACKED
tcpflags   tcp  --  anywhere             anywhere            
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request /* Ping */
ACCEPT     udp  --  anywhere             anywhere             udp spts:bootps:bootpc dpts:bootps:bootpc /* DHCPfwd */
Reject     all  --  anywhere             anywhere            
LOG        all  --  anywhere             anywhere             LOG level info prefix "Shorewall:vpn-fw:REJECT:"
reject     all  --  anywhere             anywhere            [goto] 

Chain vpn-net (1 references)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     all  --  anywhere             anywhere            

Chain vpn_frwd (1 references)
target     prot opt source               destination         
dynamic    all  --  anywhere             anywhere             ctstate INVALID,NEW,UNTRACKED
smurfs     all  --  anywhere             anywhere             ctstate INVALID,NEW,UNTRACKED
tcpflags   tcp  --  anywhere             anywhere            
vpn-net    all  --  anywhere             anywhere
̀̀̀```

Ysurac, 2018-02-21:
All seems ok here.

You should have no more *omr-tracker* errors in the system log. If you have some, wait a little, a new image is compiling and should fix this...

**Else**
You could try to run ss-server manually on the VPS: ```ss-server -c /etc/shadowsocks-libev/config.json``` (do a ```systemctl stop shadowsocks-libev-server@config``` before) and check on the console if you have something displayed when doing a ```curl ifconfig.co``` on the router.

If no, do the same on the router, ```killall -9 ss-redir```, ```ss-redir -c /etc/var/shadowsocks-libev/ss_redir.hi.json``` and ```curl ifconfig.co``` and check if there is something somewhere.

olaulau, 2018-02-22:
tried with router v0.7 vdi image this morning, still doesn't work : 

no more omr-tracker errors in syslog, but it is full of lines like this : 
```
Thu Feb 22 09:40:46 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.10.1 dev wan1
Thu Feb 22 09:40:51 2018 user.notice post-tracking-post-tracking: Replace default route by 192.168.11.1 dev wan2
```

I don't understand on which host:port I should query the `curl ifconfig.co`.

olaulau, 2018-02-22:
root@OpenMPTCProuter:~# cat /etc/config/network 
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option multipath 'off'

config globals 'globals'
	option ula_prefix 'fdae:7104:30d0::/48'
	option multipath 'enable'
	option mptcp_path_manager 'fullmesh'
	option mptcp_scheduler 'default'
	option congestion 'olia'

config interface 'lan'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option multipath 'off'
	option ip4table 'lan'

config rule 'lan_rule'
	option lookup 'lan'
	option priority '100'

config interface 'wan1'
	option proto 'static'
	option type 'macvlan'
	option ip4table 'wan'
	option multipath 'master'
	option defaultroute '0'
	option ifname 'wan1'
	option label 'crystal'
	option interface 'eth0'
	option ipaddr '192.168.10.2'
	option netmask '255.255.255.0'
	option gateway '192.168.10.1'
	option metric '1'

config interface 'wan2'
	option proto 'static'
	option type 'macvlan'
	option ip4table 'wan'
	option multipath 'on'
	option defaultroute '0'
	option ifname 'wan2'
	option label 'revolution'
	option interface 'eth0'
	option ipaddr '192.168.11.2'
	option netmask '255.255.255.0'
	option gateway '192.168.11.1'
	option metric '2'

config device 'wan1_dev'
	option name 'wan1'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1519287461'

config device 'wan2_dev'
	option name 'wan2'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1519287461'

config interface 'glorytun'
	option ifname 'tun0'
	option proto 'none'
	option ip4table 'vpn'
	option multipath 'off'
	option defaultroute '0'
```

Ysurac, 2018-02-22:
All seems ok here too. At least if your box have ip _192.168.11.1_ and _192.168.10.1_.
You should be able to ```ping 192.168.11.1 -I 192.168.11.2``` and ```ping 192.168.10.1 -I 192.168.10.2``` without problems (and without any packets lost) on the router.

**If true**, you should also be able to ```ping 8.8.8.8 -I 192.168.11.2```, on the router, even if packets are lost because of routes changes by the script.

**If true**, on the router, ```wget -O - https://github.com/Ysurac/openmptcprouter-feeds/raw/master/openmptcprouter/files/etc/uci-defaults/1940-omr-dns | sh```, this will fix a bug with DNS that is solved but not if you used a saved config. Then reboot and try on the router ```curl ifconfig.co```, this will give you your ip.

olaulau, 2018-02-22:
no, can't ping my boxes from router : 
```
root@OpenMPTCProuter:~# ping 192.168.11.1 -I 192.168.11.2
PING 192.168.11.1 (192.168.11.1) from 192.168.11.2: 56 data bytes
^C
--- 192.168.11.1 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss
root@OpenMPTCProuter:~# ping 192.168.10.1 -I 192.168.10.2
PING 192.168.10.1 (192.168.10.1) from 192.168.10.2: 56 data bytes
^C
--- 192.168.10.1 ping statistics ---
4 packets transmitted, 0 packets received, 100% packet loss
```

My boxes have been set to `192.168.10.1` and `192.168.11.1`, DHCP disabled, plugged on the same network.
They are working correctly, I'm using them with manual IP addressing on my computer (which hosts virtualbox, running the router with bridged network) : 
```
$ ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1) 56(84) bytes of data.
64 bytes from 192.168.10.1: icmp_seq=1 ttl=64 time=3.67 ms
^C
--- 192.168.10.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 3.677/3.677/3.677/0.000 ms
```

Ysurac, 2018-02-22:
OK, so it's the problem.

Your computer don't use same IPs ? You have promiscuous mode activated for virtualbox ?

olaulau, 2018-02-22:
no IP addressing conflict.
I've tried with and without promiscuous mode, which do you recommend ?

Ysurac, 2018-02-22:
with promiscuous mode.

olaulau, 2018-02-22:
```
root@OpenMPTCProuter:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.11.1    0.0.0.0         UG    0      0        0 wan2
192.168.10.0    0.0.0.0         255.255.255.0   U     1      0        0 wan1
192.168.11.0    0.0.0.0         255.255.255.0   U     2      0        0 wan2
```
strange, no ?

Ysurac, 2018-02-22:
no. No problem here, I've the same.

olaulau, 2018-02-23:
also tried with 0.7.2 images, this time under windows (I was under ubuntu), with virtualbox and vmware player.
also tried the img format on another computer, with qemu/KVM.
=> same problem : can't ping my boxes.

a friend of mine has little more success with his rPi3.
are you sure other images are working ?

Ysurac, 2018-02-23:
I'm always working on the virtualbox images. So they are working.
You can test official openwrt images: https://downloads.openwrt.org/releases/17.01.4/targets/x86/64/
If you have the same problem then it's on your side, else it's on my side (and I will have to reproduce it...).

Ysurac, 2018-02-23:
Do you have a firewall on your computer ? if yes, this may be the problem.

dougalito, 2018-02-23:
Hello,
I work with Olaulau to test your project.
I use a Pi3 and the same VPS as Olaulau.
When i used the 0.5.3 version it was nearly to work correctly. (going throw the VPN with shadowsocks but no additionnal on bandwith, seem to use only one, but my output IP is the same as the VPS).
Since i use 0.7.3 version nothing work, when i finished to configure my Pi3 i have no network on my computer and no internet. But I have an IP gave by Pi3 but nothing.
I can ping something like 8.8.8.8 but it seem to be a DNS problem. (I tried to fix it by put google DNS on my network card but still down).

Tell me if you want some LOG.

Ysurac, 2018-02-23:
This will be fixed in next release (in about 30 minutes).
Server in dnsmasq is set to _127.0.0.1_ instead of _127.0.0.1#5353_ to relay DNS to unbound.

Ysurac, 2018-02-23:
VPS script was also updated to version 0.4 that add ```"mode":"tcp_and_udp",``` in _/etc/shorewall-libev/config.json_. This allow UDP over shadowsocks, I forget this...

dougalito, 2018-02-25:
I tried few minutes ago the new release 0.8.1.
after first installation i reboot the PI3 and got this message : 

Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:7fd::1 port 53
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:503:c27::2:30 port 53
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:7fd::1 port 53
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:500:1::53 port 53
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:500:a8::e port 53
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: sendto failed: Permission denied
Sun Feb 25 18:32:05 2018 daemon.notice unbound: [1525:0] notice: remote address is 2001:500:200::b port 53

again and again.
Got an local IP from PI3 but no DNS on firefox for going to website.
can ping 8.8.8.8

VPS script is up to date

Ysurac, 2018-02-25:
Unbound notice are not a problem, only that there is no IPv6.
On the PI3 can you do ```curl ifconfig.co``` if you have the VPS ip as result, it's ok.

dougalito, 2018-02-25:
couldn't resolve host 'ifconfig.co'

and

Sun Feb 25 18:34:23 2018 daemon.err /usr/bin/ss-tunnel[2892]: [udp] server_recv_sendto: Network unreachable
Sun Feb 25 18:34:26 2018 daemon.err /usr/bin/ss-redir[2888]: fast_open_connect: Network unreachable
Sun Feb 25 18:35:07 2018 daemon.info odhcpd[635]: Using a RA lifetime of 0 seconds on eth0
Sun Feb 25 18:35:08 2018 daemon.err omr-tracker[5227]: RTNETLINK answers: No such process

dougalito, 2018-02-26:
Ok with the 0.8.2 
curl give me VPS ip.
Got a DHCP ip from PI3
Internet working Well

I just have to test if i got both connections input and output

Ysurac, 2018-02-26:
Cool :)

dougalito, 2018-02-26:
What Kind of différences betwween 0.8.2 and 0.8.3 ?

Ysurac, 2018-02-26:
In fact it's 0.8-3, only a new compilation. Nothing major: I removed example tunnel in shadowsocks.

dougalito, 2018-02-26:
At first time it seem to be ok, ...
but my connections are not reconized with their max bandwith.
My first alone Got nearly 6Mb/s and my second is about 5 Mbs test without MPTCP Router and one by one and with wy Overthebox from OVH got the good addition of it.
but on MPTCP Router one is 3.8 and the other is 2.8 and got 6 with Shadowsocks.
So i got a great difference in perf and its better for me to get only one active than both with MPTCP.
Do you have an idea about this problem ?

Ysurac, 2018-02-26:
How do get speed of each connection on the router ?

With Shadowsocks you always loose a little bandwidth because of encryption.

dougalito, 2018-02-27:
Sync stats on Freebox, and verified directly with RJ45 from the box on m'y computer, one n'y one connection.
I should try just to desactivate each connecrion on Pi3 to know if is it because of openwrite or shadowsocks

olaulau, 2018-03-17:
HI ;

Just tried last build on VPS and on an rPi3, works pretty great !
I'll try again from virtualbox later, and let you know in the other ticket.
perfs are not as good as expected, especially concerning upload and ping (download is good). I'll open a ticket later with more infos.
I think we can close this ticket.

thanks for the good work, looks very promising.
Laurent.

closed 2018-03-17 by olaulau

olaulau, 2018-04-08:
just a small message to tell you that last versions seems to work well,  I'm seeing huge improvements in term of download speed and responsiveness in web browsing.
great job ! thank you.