Ysurac/openmptcprouter #2: Whatsapp call isnt functionable

Liixee, 2018-02-22:
Hi friend,

great project, but i have a problem with the whatsapp call.
Whatsapp can't connect with the other call-partner.

Does UDP work over MPTCP and SS?

Ysurac, 2018-02-22:
UDP work over shadowsocks without problems.

You may have to open some ports on shorewall on the VPS.

I can't test: whatsapp seems to block my VPS (OVH) even for account creation and there is no voice call test.

Ysurac, 2018-02-22:
You can also try to stop shorewall on the VPS at least to try if this is a port problem: ```systemctl stop shorewall```

Liixee, 2018-02-22:
Okay, i have test it but the problem is not solved. You can test it with ipferf.

Liixee, 2018-02-22:
I have in 
/lib/systemd/system/shadowsocks-libev-server@.service the 
ExecStart=/usr/bin/ss-server -c/etc/shadowsocks-libev/%i.json
chanched in:
ExecStart=/usr/bin/ss-server **-u** -c/etc/shadowsocks-libev/%i.json

and udp is working.

Ysurac, 2018-02-22:
Thanks ! I forget this one...
You can add ```"mode": "tcp_and_udp"``` in ```/etc/shadowsocks-libev/config.json``` instead of the -u.

Whatsapp is working now ?

Liixee, 2018-02-22:
Yes, Whatsapp is working now :)

Ysurac, 2018-02-22:
Cool thanks.
I fixed the VPS script.

closed 2018-02-22 by Ysurac

talalmash, 2020-04-26:
> Thanks ! I forget this one...
> You can add `"mode": "tcp_and_udp"` in `/etc/shadowsocks-libev/config.json` instead of the -u.
> 
> Whatsapp is working now ?

I still have to do this on version: 0.1015 4.19.104-mptcp in order to get UDP working. 
Also set Redir Rules > ssd-redir for UDP to hi2 - tcp_and_udp instead of hi , though this is on an unofficial build client/router. Does the client configure this remotely?