Ysurac/openmptcprouter #4: can't ping boxes from routeur

olaulau, 2018-02-28:
Hi ;

I tried lede-17.01.4-x86-64-combined-ext4.img,
changed /etc/config/network config to 192.168.10.2 ,
then I could ping 192.168.10.1 .

same VM but with openmptcprouter-v0.9.1-1-g4af41a2-v0.9-9-g6715b51-x86-64-combined-squashfs.img,
made the config like the documentation said,
can't ping 192.168.10.1 .

sadly the problem isn't on my side.
any idea ?

Liixee, 2018-03-01:
Firewall rules checked?

Ysurac, 2018-03-01:
From a fresh image, only changing on the interface wan1 IPv4 address to 192.168.10.2, IPv4 netmask to 255.255.255.0 and IPv4 gateway to 192.168.10.1, then "Save & Apply". Then on the image by SSH or using Diagnostics I can ping the gateway 192.168.10.1

closed 2018-03-29 by Ysurac