Ysurac/openmptcprouter #7: IPv6 support

Ysurac, 2018-03-07:
Add full IPv6 support

Ysurac, 2018-03-12:
In latest commit, basic IPv6 support enabled.
LAN get ULA IPv6, can connect to IPv6 TCP & UDP server using VPS IPv6.
Need to add:
* redirect port from IPv6 VPS to any IPv6 on home network
* public IPv6 available for home network in both side

Ysurac, 2018-03-14:
Problem on RPI3
```
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.401816] CPU: 1 PID: 14 Comm: ksoftirqd/1 Not tainted 4.9.82 #0
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.409718] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.417239] Call trace:
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.421261] [<ffffff80080852d4>] dump_backtrace+0x0/0x180
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.428281] [<ffffff8008085468>] show_stack+0x14/0x1c
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.434897] [<ffffff80082149b8>] dump_stack+0x8c/0xac
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.441453] [<ffffff8008369f2c>] netdev_rx_csum_fault+0x38/0x44
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.448861] [<ffffff800835e834>] __skb_checksum_complete+0x6c/0xb8
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.456497] [<ffffff80083b7ad0>] tcp_rcv_established+0x464/0x578
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.463919] [<ffffff8008420828>] tcp_v6_do_rcv+0x180/0x418
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.470763] [<ffffff800842160c>] tcp_v6_rcv+0xb4c/0xe24
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.477288] [<ffffff80083fdf58>] ip6_input_finish+0xcc/0x358
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.484214] [<ffffff80083fe214>] ip6_input+0x30/0x80
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.490374] [<ffffff80083fde6c>] ip6_rcv_finish+0x8c/0xac
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.496928] [<ffffff80083fe4fc>] ipv6_rcv+0x298/0x308
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.503074] [<ffffff80083695f0>] __netif_receive_skb_core+0x668/0x838
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.510615] [<ffffff800836a238>] __netif_receive_skb+0x2c/0x78
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.517478] [<ffffff800836b31c>] process_backlog+0x90/0x134
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.524058] [<ffffff800836b108>] net_rx_action+0x104/0x288
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.530537] [<ffffff8008094700>] __do_softirq+0x10c/0x204
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.536913] [<ffffff8008094824>] run_ksoftirqd+0x2c/0x50
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.543166] [<ffffff80080ac1ec>] smpboot_thread_fn+0x150/0x174
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.549935] [<ffffff80080a8c68>] kthread+0xd8/0xe0
Wed Mar 14 11:22:33 2018 kern.warn kernel: [ 1241.555620] [<ffffff8008082330>] ret_from_fork+0x10/0x20
Wed Mar 14 11:22:33 2018 kern.err kernel: [ 1241.562060] <unknown>: hw csum failure
```

I investigate to find where is the problem exactly.

Ysurac, 2018-03-14:
No more issue if using ```ethtool --offload eth0 rx off tx off```

Ysurac, 2018-03-15:
After some time, this make RPI crash...
I will try latest MPTCP 0.93 from git.

lars18th, 2018-04-03:
Hi @Ysurac ,

Please, remember to leave the option for completly disable IPv6 if you add support for it!
I hate to see IPv6 traffic in my LAN & ROUTERS as the security can be compromissed if you don't configure it properly.

If it's a _global_ option then no problem at all! :smile:

Ysurac, 2018-04-03:
IPv6 support is added in OpenMPTCProuter since 0.11, but **Always announce default router** in interface IPv6 settings must be activated to make it work. It's only ULA IPv6 so no real security problems here.

lars18th, 2018-04-03:
> but Always announce default router in interface IPv6 settings must be activated to make it work

OK. But is not possible to have a global option for disabling it?

> It's only ULA IPv6 so no real security problems here.

The problem with IPv6 is when unconfigured devices start to forward IPv6 traffic and you don't know! And this it's very easy to happen. So, for this reason I prefer to disable it in all of my routers.

I hope isn't difficult to support (disable) it. :smile:

Ysurac, 2018-04-03:
If the route is not announced, you don't use it. So why disabling something that is not used and don't really change anything ?

It's the same with IPv4, devices must be configured. But when route is not announced, nothing will use it.

lars18th, 2018-04-03:
> If the route is not announced, you don't use it.

Sorry for my misunderstanding.
If you don't announce a _default gateway_ in the IPv6 interface of the Open-MPTCP-Router then all IPv6 traffic that arrives to that interface is never used (allways discarded)?
If this is true, then no problem. If not, then we can be in a trouble.

The problem is that a lot of devices can today automatically discover IPv6 gateways, and this can be a security hole. That's the reason for disabling IPv6 in all of my routers.

Ysurac, 2018-04-03:
No, if you set the gateway manually this will be used. It's a router, it route anything that come :)
You can disable **Router Advertisement-Service** and **DHCPv6-Service** in lan interface, then no way to discover anything.

You can also disable IPv6 traffic using OpenMPTCProuter firewall.

lars18th, 2018-04-03:
> No, if you set the gateway manually this will be used.

And this **is** the problem!

> You can disable Router Advertisement-Service and DHCPv6-Service in lan interface, then no way to discover anything.

Searching (testing) for a IPv6 GW is not very difficult. And if your firewall/router rules aren't configured for IPv6... then you **really** be in a trouble!

> You can also disable IPv6 traffic using OpenMPTCProuter firewall.

Then I suggest to make this config in UI very **VISIBLE**.
But, in any case I prefer one option for disabling IPv6 globally (for example, with a sysctl entry).

Ysurac, 2018-04-03:
* It's a router. If it route traffic then it's not a problem.
* If you have on your internal network really strange device that scan all ULA IPv6 network (and have lot of times for that), maybe. But in this case you should really buy another device ;)
By default firewall for OpenMPTCProuter VPS and OpenMPTCProuter is configured for IPv4 and IPv6 the same way.
* I will not change anything on the firewall. You can use sysctl in /etc/sysctl.d and I will also add a sysctl UI for some advanced change (issue #30)

lars18th, 2018-04-03:
> It's a router. If it route traffic then it's not a problem.

It is when you configure properly the IPv4 part and leave with defaults in the IPv6 part.
Please, think that I don't like to force anything. I only like to suggest to leave the user the option for disable IPv6 if s/he doesn't likes to use it. Nothing more. How to achieve it is not relevant.

> I will not change anything on the firewall. You can use sysctl in /etc/sysctl.d and I will also add a sysctl UI for some advanced change (issue #30)

Perhaps a simple note like "For disabling IPv6 insert: net.ipv6.conf.all.disable_ipv6 = 1; net.ipv6.conf.default.disable_ipv6 = 1; net.ipv6.conf.lo.disable_ipv6 = 1" in this advaced section will be sufficient.

Regards!

Neustradamus, 2019-04-30:
Some French providers change and use now IPv6 by default, can you look it?

GermanAizek, 2024-09-03:
@Ysurac does omr-vps already have ipv6 support? and will omr-router be able to connect to it over ipv6?

GermanAizek, 2024-09-04:
@Ysurac please answer in this thread.

Ysurac, 2024-09-04:
Old issue...
OMR VPS part have IPv6 support, and you can connect to it over IPv6. To reach IPv4 websites you still need to have an IPv4 on the server or DNS64/NAT64.