Ysurac/openmptcprouter #8: Add a wizard

Ysurac, 2018-03-07:
To help configure OpenMPTCProuter create a wizard.

Ysurac, 2018-03-31:
Some features needed:
* Add interfaces in wizard
* Detect if DHCP is on for modems ?
* Check if modems are not on the same network

Ysurac, 2018-05-15:
Interfaces support added
Wizard can be better with detection if all settings will work.

rstanislav, 2018-08-07:
Why there is need for static address on modems ? With DHCP leased it will not work ? Most modems works in HILINK mode, that means that they are discovered as network card, and if there is more than 1 modem plugged there is always chance that if port changed on something other happens - it will mess device naming inside, so eth1 becomes eth2, eth2 - eth1, and with static address everything messed...
Also wizard does not store port numbers for shadowsocks and glorytun, so after save\apply it resets to default values =(

Ysurac, 2018-08-07:
It can work with DHCP for USB modems or if each network modem is put on different Ethernet ports. I will add DHCP support in the wizard.

All modems are not USB and many modems doesn't support HILINK mode. It's not possible to support every possible configurations, and I have no USB modems for tests.

Wizard is mean to work with OpenMPTCProuter  VPS for basic users, users that change ports of Shadowsocks and glorytun are not the target of it.

closed 2018-09-21 by Ysurac