Ysurac/openmptcprouter #12: Problème DNS et Lag

Kuma68, 2018-03-12:
Bonjour,

Je tiens tout d'abord à vous remercier pour le travail effectué !

Je revend les solutions OverTheBox, mais depuis les nouveaux tarifs je cherche une autre solution..

J'ai testé le Bonding VPN en RR et c'est fonctionnel mais avec le MPTCP j'ai clairement un meilleur débit !

Malgré les performances, j'ai quelques bug : 
- Impossible d'utiliser des DNS custom (8.8.8.8), il faut obligatoirement passer par les DNS OpenMptcpRouter
- J'ai certains site qui lag, pour exemple www.megavoip.com

Mon infra (j'ai essayé de faire complet et compréhensible): 
Internet --> Dédié SoYouStart --> ESXI 6.0 --> pfSense (NAT) --> Debian <-- VPN OPENMPTCP --> OpenMptcpRouter --> pfSense (WAN) --> ESXI 6.5 --> PC

Sur site j'utilise une freebox et une sfrbox en mode routeur, dhcp désactivé.
J'utilise une IP dédié pour MPTCP (entrant et sortant), la VM est dédié également.
Depuis une VM sur le SoYouStart, www.megavoip.com fonctionne très bien.

Merci de votre aide !

Si jamais il est possible de vous aider au développement (Traductions, tests...), je serais heureux de participer !

Ysurac, 2018-03-12:
Il me faut les versions d'**OpenMPTCProuter** et du script **OpenMPTCProuter VPS** utilisées.

Sur mes propres tests:
* Aucun soucis pour moi. Que voulez vous faire ? annoncer d'autres DNS ? un dig @8.8.8.8 google.com sur un PC client ne fonctionne pas ?
* Aucun lag pour moi. Après vu votre config il peut y avoir des soucis de MTU.

Kuma68, 2018-03-12:
Bonjour,

J'ai fais l'install hier, donc les ce sont les dernières versions : 
- openmptcprouter v0.10-5-g2f4e191 r6328-d8655868ca / LuCI build branch (git-18.066.46907-2f4e191)
- OpenMPTCProuter VPS -> Comment savoir la version ?

J'ai pour habitude d'utiliser les DNS google car je change souvent de gateway.
Je viens de faire un nslookup : 

```
C:\Users\Michael>nslookup
Serveur par dÚfaut :   UnKnown
Address:  192.168.1.250

> google.com 8.8.8.8
Serveur :   [8.8.8.8]
Address:  8.8.8.8

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
*** Le délai de la requête sur 8.8.8.8 est dépassé.
> google.com 192.168.1.250
Serveur :   [192.168.1.250]
Address:  192.168.1.250

Réponse ne faisant pas autorité :
Nom :    google.com
Addresses:  2a00:1450:4007:805::200e
          172.217.18.206
```

En passant directement via OpenMptcpRouter en local j'ai l'impression d'avoir moins de lag sur certains sites, mais pas tous. Le site de la banque populaire (une fois logué) est quasi impossible à consulter..
Il y a un moyen de diagnostique pour trouver la cause ?

Merci.

Ysurac, 2018-03-12:
* Ce n'est pas la dernière version, j'ai sorti la 0.11.
* La version est visible dans /etc/motd, et normalement à chaque fois qu'on se connecte en SSH au VPS

Les DNS par défaut sont les root serveurs afin d'éviter l'utilisation des serveurs de google justement :)
Il faudrait vérifier si ss-server sur le VPS écoute bien en TCP et UDP (avec netstat par exemple) et pareil sur OpenMPTCProuter.

TCPdump peut permettre de diagnostiquer, mais je commencerais par mettre un MTU inférieur à 1500 pour shadowsocks par exemple.

Ysurac, 2018-03-12:
Il faudrait aussi vérifier le TCP congestion sur le VPS et OpenMPTCProuter, ça peut jouer parfois.

Kuma68, 2018-03-20:
Bonjour,

Je suis passé à la 0.11 depuis quelques jours et les différents problèmes ci-dessus sont résolu.

J'ai par contre maintenant des petits problèmes de stabilité, est-il possible de monitorer shadowsocks (état de la connexion, paquets perdu...)
J'ai de temps en temps le lien qui tombe, j'ai déjà du reboot pour qu'il ce stabilise.

Le soucis c'est que j'ai une appli (Ciel Gestion Commercial Cloud) et elle est très sensible à la connexion, dès qu'il y a une micro coupure elle crash..

Ysurac, 2018-03-20:
Pour les mises à jour pour le moment il n'y a pas vraiment de solution. C'est prévu, mais pas pour tout de suite. On peut cependant sauvegarder la configuration et la réappliquer mais parfois certaines nouveautés dans la configuration peuvent être effacé par la sauvegarde.

Il est compliqué de monitorer shadowsocks, c'est prévu mais il faut que je trouve la solution la meilleure.

J'ai également ce soucis de coupure de temps en temps, j'ai identifié une partie qui sera réglé lors du passage du noyau en 4.14, pour le reste il faut que je trouve la source du soucis.

Kuma68, 2018-03-30:
Bonjour,

J'étais passé à la 12.1 et j'avais toujours des problèmes avec shadowsocks alors que glorytun fonctionnait plutôt bien.
Je suis passé à la 14.1 hier et pour l'instant je trouve la stabilité bien meilleur mais pas au top, mon appli crash toujours, elle est vraiment pénible x)
Le graphique MPTCP est le bienvenu également !

Un autre élément qu'il manque et qui est très bien fait dans les routeurs Draytek, c'est la gestion de la bande passante par @IP / @MAC, le top serais :
- Un graphique en temps réel de la BP (up et down) utilisé par une IP ou MAC
- La possibilité de brider une IP ou MAC à une certaine BP
- La possibilité de bloquer le traffic vers internet d'une IP ou MAC

Ysurac, 2018-03-30:
Quand une connection est inactive au bout d'un certain temps elle est tuée. Il est possible de changer ça en modifiant ```sysctl -w net.ipv4.tcp_keepalive_time=ladurée```, c'est modifié à 2400 pour OpenMPTCProuter et c'est à 7200 par défaut sous Linux. Une valeur plus basse permet avec MPTCP d'éviter d'avoir pleins de connections inutiles ouvertes, c'est souvent modifié à la baisse ainsi sur les routeurs. Il faudrait voir si l'application permet sinon de régler un keepalive...

Pour la gestion BP, je verrais si j'ajoute ces options ou si un paquet pour OpenWrt existe déjà.

PxxBxx, 2018-07-27:
Bonjour, je me joins à cette discussion, mais soit dit en passant, c'est extrêmement prometteur, j'ai réussit à faire fonctionner depuis 2j (presque) tous mes accès TCP au travers de Shadowsocks, et ce, avec jusqu'à 3 WAN.

* Setup : 
  * VPS en v0.37 (Debian9.5 chez OVH, interface en eth0 (et non ens3))
  * Routeur RPI3 en v0.29
  * wan1 en 192.168.1.0/24 (routeur 4G/ethernet Free)
  * wan2 en 192.168.3.0/24 (adsl pitoyable Orange)
  * wan3 en 192.168.2.0/24 via un usb-modeswitch (routeur portable 4G/wifi de Bytel)

* Problème(s) : 

  * Impossible d'accéder à un DNS autre que celui du routeur. Or une Chromecast utilise "en dur" les DNS de google.
Je n'ai touché à aucune configuration sur le routeur (v0.29) et le VPS est également out of the box (v0.37)
Exemple : 
```
dig @8.8.8.8 google.com

; <<>> DiG 9.8.3-P1 <<>> @8.8.8.8 google.com
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached
```
  * Je n'arrive pas à faire passer les appels ICMP au travers du router.
Et ce, quel que soit le VPN choisi (Glorytun TCP ou UDP, je n'ai pas essayé OpenVPN).
Je soupçonne un problème de shorewall, mais je ne maitrise pas son usage, et les règles en place (VPS) sont celles du script en v0.37.
Depuis un client (linux ou osx) : 
```
PING 8.8.8.8 (8.8.8.8): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
```
Trace coté VPS (syslog) : 
  * Glorytun TCP
```
glorytun-tcp-run[506]: xx.xx.xx.xx.32632: connected
kernel: [ 4562.350656] IPv4: martian source 8.8.8.8 from 192.168.100.128, on dev gt-tun0
```
  * Glorytun UDP
```
kernel: [ 4501.297061] IPv4: martian source 8.8.8.8 from 192.168.100.128, on dev gt-udp-tun0
```

Merci d'avance pour ce super projet, je viens de déménager à la campagne et doit télétravailler, l'ADSL 1.5Mbps est insuffisant, et une simple 4G de qualité n'est pas assez stable. Mais via SS+MPTCP les premiers résultats sont plus qu'encourageants.

Ysurac, 2018-07-27:
Ca a aucun rapport avec le bug déjà ouvert. Merci d'ouvrir un nouveau bug.

PxxBxx, 2018-07-28:
pardon @Ysurac le début du thread mentionnait le même problème d'accès DNS avec une version <0.11.
Mais cette partie du problème étant résolue pour @Kuma68 en 0.11, vous avez raison, autant ouvrir une nouvelle issue, ce que je viens de faire.
En tous cas, continuez le boulot, ça fonctionne quasiment out of the box vos packages VPS et Router, et en TCP, j'ai des résultats plus qu'agréables.

Cmoi20, 2018-08-23:
Bonjour,
peut être que cela peut aider pour la mise en place d'un graphique de bande passante par IP : https://github.com/wtc0der/wtBRMon

closed 2018-11-26 by Ysurac