Ysurac/openmptcprouter #13: Add choice to remove encryption

Ysurac, 2018-03-12:
Add choice of none encryption for shadowsocks and glorytun.

Ysurac, 2018-03-13:
This need to patch shadowsocks and create a deb or compile from source for the VPS part.

nileio, 2018-07-04:
how to disable encryption for both glorytun and shadowsocks please? where are those options ?

Ysurac, 2018-07-04:
For ShadowSocks it's in ShadowSocks interface (Remote servers->Method->None). Same method must be used on VPS (/etc/shadowsocks-libev/config.json).
It's not possible for Glorytun.

nileio, 2018-07-04:
thanks,
for shadowsocks
i found it on the VPS , method: aes-256-cfb' -> change this to none
	
glorytun:
there is a tick box for chacha20 on glorytun from the interface, and also glorytun tun0 includes OPTIONS="chacha20 multiqueue keepalive" .. can i remove chacha20 ? will this disable encryption on glorytun ? can you clarify

Ysurac, 2018-07-04:
If you disable chacha20 this use aes256gcm.
But really I don't know any good reason to disable encryption of ShadowSocks and Glorytun.

nileio, 2018-07-04:
i am trying to test whether i will gain any bitrate differences by disabling encryption

nileio, 2018-07-04:
i disabled encryption on shadowsocks but not sure what it is my internet connection is unstable.. many sites timeout or give socket errors !!

Ysurac, 2018-07-04:
Encyption overhead is really small, you will not get a big bitrate difference.
Did you restart ShadowSocks on VPS ?