Ysurac/openmptcprouter #17: Make speed tests and optimizations

Ysurac, 2018-03-17:
Need to make a lot of tests and optimizations for TCP. Shadowsocks optimizations doesn't seems to be enough.
* All IPv4 TCP congestions must be tested (Not sure OLIA is always better than LIA).
* Tests with and without MPTCP checksum
* Tests all shadowsocks encoding methods (rc4-md5, chacha20,...)
* Try to optimize some TCP settings

All tests will be made on MTCP 0.94.

Ysurac, 2018-03-30:
Some speed tests with IPv4 TCP Congestion and MPTCP:
* BBR doesn't seems to give a good result
* CUBE doesn't give a good result too
* LIA and OLIA seems to give close results, this need more tests.

Ysurac, 2018-03-30:
WVEGAS is not so bad but slower tha OLIA in my test.

uryupinsk, 2019-05-13:
Regarding encryption methods for both shadowsocks and glorytun, AES is the best choice in terms of speed for AES-NI compatible CPUs. It would be great to switch automatically the encryption method on the VPS side when it's modified on OpenMPTCProuter.

Any advancements on congestion control methods? As of today, it seems that you chose BBR. Is it really the best choice for all scenarios? If not, we could test it at home, as users here are in a lot of different configurations, and see what method is best for a given connection type. Maybe with an unified method?

Ysurac, 2019-05-13:
Encryption method is already switched on VPS side when modified on OpenMPTCProuter.

BBR seems to be faster in most scenarios for now. But I'm open to other tests ;)

uryupinsk, 2019-05-14:
Oh, didn't know that! It has not been very hard to modify the files by hand but it is a nice little thing to have.

I'm open to it too. What do you suggest to do for testing it?

user747, 2019-12-16:
do you think it would be a good idea to have some testing scripts? for protocol only testing, i think a omr-headless version would be interesting. record data and see how snapshots perform. maybe too early since still in alpha stage. 

this would be done on user side ( unless there are server resources for this) 

run everything in QEMU ( maybe inside of something like gns3-server) , set up a few virtual wans, see what happens.

gnosoz, 2020-04-24:
I am thinking of running an extensive test (cisco trex, solarwind, etc) to simulate a series of scenarios.
in particular I was think of the following stress tests:
      - maximum concurrent sessions;
      - simulate heavy udp traffic up/down;
      - simulate heavy tcp traffic up/down;
      - simulate users (1-50) generic traffic (mail, web, Skype, youtube, SMS, gaming)  
       - simulate sip phones (1-50);
       - simulate mobile phones (1-50);

Furthermore I will test to see if I can reach 3Gbps router speed if inoculated with 3 x 1Gbps WANs. 

All tests to be run with 1 to 5 WANS and all combinations of schedulers/congestion.

Let me know if you have any suggestione on additional tests to performe and tools so we can try standardize the process and add finding from all users.

I will publishing my finding here.

SlipstreamCoding, 2020-04-30:
I'm really eager to see the reults of these tests.
It would be interesting if you could simulate ADSL, 4G/LTE fluctuations in the mix. Maybe even have a test scenario for a heavily congested situation with links failing and coming back ?

gnosoz, 2020-05-02:
> I'm really eager to see the reults of these tests.
> It would be interesting if you could simulate ADSL, 4G/LTE fluctuations in the mix. Maybe even have a test scenario for a heavily congested situation with links failing and coming back ?

Funny you should say that as I have few 4G dedicated routers and the fluctuations are absolutely real, no need for simulations of those hahahah.

the falling and coming back is done regularly by unplugging the cables.

the lines go from 700Kbps to 7/12Mbps download, by night is even worse let me tell you... 
On the plus side I can start a new Netflix competitor as the upload I get is amazing, all these people at home use zero upload ;)

At the moment I need to complete the configuration of the platform to monitor everything and I am using Nagios core + Munin. 

muninlite (the openwrt version of munin)  in all but one install failed miserably to start so I am now edging towards either SNMPD direct call from munin/nagios or ssh to execute muninlite...

ssh might change the results as munin starts several concurrent calls and that might weight the router down too much to give reliable cpu/ram  results.

SlipstreamCoding, 2020-05-03:
I'm not a hardcore sysadmin so I can only nod at the nagios/munin/snmp part, though I know enough to feel you'll get some nice graphs out of them :) 

On my part I'm hoping to get the cleanest, most stable connexion for portable video streaming (upload), wardriving style. 
Will you be monitoring jitter as well as throughput? Any graphing packages to recommend for openwrt? 

The proof of concept works on raspi3b with 4 USB modems, but I need more robust and am tinkering with a Mitac SBC 3.5", pcengines APU, mpcie modem, custom lifepo4 UPS... To integrate into a pelicase with custom plate. 
Maybe it'll be a raspi4b in the end? 
Antenna system is still not elected. 

#stayathome is enabling this development, but also making it too slow to my taste. 
Hopefully your research will speed things up for me when the hardware side of things is figured out, and can finally start testing IRL. 
Looking forward to it !

talalmash, 2020-05-27:
Needs better UDP VPN performance (latency mostly) when the TCP side is congested.

sadahsan, 2021-11-12:
> I am thinking of running an extensive test (cisco trex, solarwind, etc) to simulate a series of scenarios. in particular I was think of the following stress tests: - maximum concurrent sessions; - [...]

Hi, did you manage to complete your testing? What was your findings mainly interested in aggregated throughput speed

mstrolo, 2022-01-27:
Same here would like to know