Ysurac/openmptcprouter #18: QoS : best ping for mall packets

olaulau, 2018-03-17:
It would be great to have a small packet detection that automatically categorize these traffic as (ping) high priority.
It should also detect, at any time, which of the WAN has the best ping to serve those connexions.

Actually, it seems that event ping packets are round-robined. It's easy to see that because my 2 DSL connection doesn't have the same base ping.

Ysurac, 2018-03-17:
Ping (ICMP) is not using ShadowSocks tunnel (and ping is always low priority).
ShadowSocks is only used for TCP and UDP.

I may use the glorytun VPN to redirect ping but this would be only for isolation.

olaulau, 2018-03-17:
ok so how does ping traffic transit through openmptcprouter ?
my feature request was not for ping, but mainly for other services (game, voip ...) that mainly use small packets and which depend mainly on ping for their QoS.
ping should not be low priority as some applications rely on it to test responsiveness of a service.

Ysurac, 2018-03-17:
Ping should use default route (so the interface defined as master for multipath if up)

Ping is not "real" traffic, it's not a good way to measure RTT. Many ISP give high priority to HTTP, SIP,... 
some are blocking ping. QoS doesn't use ping. I don't know any applications that rely on ping to test responsiveness and only on that.

For SIP, or anything you want, you can set QoS on OpenMPTCProuter.

olaulau, 2018-03-17:
yes by specifying ports.
but OverTheBox has a small packet detection that automatically and a best ping detection to make sure you have the best possible ping in game (for example).

Ysurac, 2018-03-17:
I don't know what OverTheBox do. Never used it.
QoS is only used when bandwidth is saturated.
Ping, ICMP echo request,  is not used by games to measure RTT. Data exchanged by games are not so small, it's not some small "echo request".

olaulau, 2018-03-17:
not real ping of course. but they manage to make some kind of ping through their own protocols during games, mostly relying on UDP.
It would be great to have some autodetection for those kind of traffics (games, voip ...).

Ysurac, 2018-03-17:
For SIP, yes I will add this to default configuration but it will by port.
For games, it's by ports too. I will add steam to default configuration too.
But it's only QoS, so this only change something when bandwidth is saturated.

olaulau, 2018-03-17:
for future enhancements, maybe think of a daemon that monitor interfaces's ping to choose the best way.
and we will try to find some rules to determine this traffic.

Ysurac, 2018-03-17:
It's already what is doing MPTCP with default scheduler.
For rules, this will always be by ports. And for Steam I already have them :)

Ysurac, 2018-03-19:
In fact, small UDP packets are already high priority by default QoS OpenWrt rules.

closed 2018-04-06 by Ysurac