Ysurac/openmptcprouter #19: range port forwarding (DMZ)

olaulau, 2018-03-18:
single port forwarding works well. I can easily chat using netcat on both sides.

## Expected Behavior
all ports from the range should be forwarded as it work with single port forward

## Actual Behavior
range forwarding doesn't work at all. instead, it seems that single port previously forwarded remains active.

## Steps to Reproduce the Problem

  1. set port to "1-64999" for external and internal side

## Specifications

  - OpenMPTCProuter version: v0.12-1-gfe1fbe2
  - OpenMPTCProuter VPS version:
  - Platform: VPS : debian 9 / client : ubuntu

Ysurac, 2018-03-18:
If it's port forwarding from the VPS to OpenMPTCProuter, it's working for me.
I will add some better default configuration, and see what I can do about SNAT. I will also add a documentation about that.

If it's port forwarding from OpenMPTCProuter to a client, never tested to redirect a full range.

Redirecting a range or a port shouldn't really change anything for configuration.

olaulau, 2018-04-08:
just tried again with lastest version of VPS & router, seems to work perfectly.
will try tomorrow with my NAS to see if all works well.

olaulau, 2018-04-08:
seems to work with my NAS.
easily configured with static DHCP !
thanks.

closed 2018-04-08 by olaulau