Ysurac/openmptcprouter #21: Pas d'accès à Internet avec la configuration par défaut

Adrien54-FR, 2018-03-20:
Bonjour,

Pour tester OpenMPTCPRouter, j'utilise une configuration un peu particulière, pour ensuite l'intégrer chez moi : 
Côté VPS, le serveur est sur un ESXI derrière un PFSense 
Côté routeur, il est dans une machine virtuel (IP : 192.168.1.1 côté WAN 1 et IP : 192.168.100.1 côté LAN)
Poste de test: une machine virtuel aussi (IP : 192.168.100.50)

Internet <=> Kimsufi ESXI 5.5 <=> PFSense <=> Debian 9 <-- VPN OPENMPTCP --> OpenMptcpRouter <=> PC sous une machine virtuelle

Lorsque seulement le shadowsock est activé, si je fais un diagnostique dans le menu sur l'interface Web, je n'ai pas l'IP du Kimsufi, j'ai celle de ma box et j'ai bien Internet sur le poste Client, le routeur 192.168.100.1 rebondi sur la Livebox 192.168.1.1
Si j'active le glorytun, j'ai bien l'IP de mon Kimsufi.
Par contre lorsque je teste sur le poste client je n'ai pas Internet... 
Le ping ne passe pas : 
`Envoi d’une requête 'ping' sur google.fr [172.217.8.3] avec 32 octets de données :
Réponse de 192.168.100.1 : Impossible de joindre le port de destination`

La passerelle 192.168.100.1 ping bien

Si je fais le test directement sur le routeur avec l'outil de diagnostique ou en ligne de commande ça fonctionne bien (commande curl et ping)

Pouvez-vous m'aider à comprendre le problème ?

Merci d'avance

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.12.1-3
  - OpenMPTCProuter VPS version: v0.9
  - Platform: Debian 9

Ysurac, 2018-03-20:
Quel est l'IP de la livebox et celle du port WAN1 ? elles doivent être différentes.
Pour ShadowSocks il faut vérifier que la clef est bien entrée et correspond à celle du VPS, que le chiffrement est également le même, et vu que PFSense est utilisé que les ports sont ouverts. Avec uniquement shadowsocks de lancer l'IP depuis OpenMPTCProuter doit être celle du VPS.

Adrien54-FR, 2018-03-21:
Bonjour,

Finalement, en remettant tout à zéro et en refaisant tout à tête reposée ça fonctionne correctement !
Je n'ai pas encore testé la redirection de port, c'est la prochaine étape..
Pour répondre à la question, les IP de la livebox et du WAN1 étaient bien différentes, et les ports bien ouverts sur PFSense.

Merci

closed 2018-03-26 by Ysurac