Ysurac/openmptcprouter #22: Résolution DNS K.O depuis version 0.12.1-8

xslite, 2018-03-25:
Bonjour, et merci pour votre excellent travail !

Donc, depuis la toute dernière version routeur client 0.12.1-8, le service de résolution DNS embarqué par défaut ne fonctionne plus en standard (après installation et configuration des WAN et Remote SS)

J'ai testé en downgradant vers 0.12.1-3 : tout fonctionne parfaitement bien dès le démarrage des instances WAN (ce qui n'est plus le cas en 0.12.1-8).

Merci d'avance pour les correctifs

  - OpenMPTCProuter version: 0.12.1-8
  - OpenMPTCProuter VPS version: 0.11
  - Platform: X64

Bien cordialement,

XSLite

xslite, 2018-03-25:
Problème identique constaté sur 0.12.1-7
dernière version fonctionnelle "out of the box" : 0.12.1-3

Ysurac, 2018-03-26:
En fait tout est bloqué, une erreur dans la création d'une régle du firewall.
La prochaine version corrigera ce soucis.

Ysurac, 2018-03-27:
Normalement c'est réglé dans la version 0.13

closed 2018-03-27 by Ysurac

xslite, 2018-03-27:
Oui, bravo, j'ai testé ce matin, ça marche du tonnerre !

Et plus de warning dans les logs au sujet de Spectre, ça fait plaisir :-)