Ysurac/openmptcprouter #23: Vue de synthèse des graphes de traffic

xslite, 2018-03-26:
Hello,

une "live view" aggrégée des traffics provenant des différents WAN serait bien pratique !

Cela permettrait en un coup d'oeil de voir si tout fonctionne bien, et comment est réparti le flux entre les box/clés 4G.

[image]
[image]
[image]

Ysurac, 2018-03-26:
C'est prévu.

Ysurac, 2018-03-28:
Un graph est ajouté dans le menu **Réseau->MPTCP->Bandwidth**.

closed 2018-03-28 by Ysurac