Ysurac/openmptcprouter #25: baux statics DHCP

Cmoi20, 2018-03-26:
## Expected Behavior
Un autre produit que j'utilise actuellement permet de spécifier dans les baux DHCP une passerelle spécifique pour forcer l'utilisation d'un WAN spécifique. cette fonction m'est aujourd'hui très utile.

## Actual Behavior
la gestion des baux DHCP pour préciser l'adresse IP est déjà fonctionnelle.

## Steps to Reproduce the Problem

  1. aller sur Réseau -> DHCP et DNS
  2. ajouter un bail statique
  3. une case permettant de préciser une passerelle serait une super avancée.

## Specifications

  - OpenMPTCProuter version: 0.12.1-3
  - OpenMPTCProuter VPS version: 0.11
  - Platform: macosx

Ysurac, 2018-03-26:
C'est une amélioration qui pourrait être faite au niveau d'OpenWrt, ce n'est pas spécifique à OpenMPTCProuter. Dnsmasq qui est utilisé pour la partie DHCP supporte l'ajout d'une passerelle par baux.

Ça permettrait également l'utilisation d'un tunnel spécifique pour une machine.

Ysurac, 2018-03-29:
Dispo dans OpenMPTCProuter 0.14 (même si je n'ai pas encore eu le temps de vraiment tester)

closed 2018-03-29 by Ysurac