Ysurac/openmptcprouter #26: inactive TCP connections are dropped

olaulau, 2018-03-29:
## Expected Behavior
like when using direct WAN, inactive TCP connections (such as SSH with keepalive activated) should not disconnect.

## Actual Behavior
inactive SSH sessions are quickly disconnected, I would say 2 minutes.

## Steps to Reproduce the Problem

  1. open a SSH session to an outside host which has keepalive active
  2. don't do anything on this terminal for a few moment
  3. the connection disconnects

## Specifications

  - OpenMPTCProuter version: v0.12-1-gfe1fbe2
  - OpenMPTCProuter VPS version: 0.8
  - Platform: rPi 3 / Debian 9

Ysurac, 2018-03-29:
You should update OpenMPTCProuter and OpenMPTCProuter VPS to latest versions.

Inactive TCP connections are disconnected to don't have too many opened connections between the VPS and OpenMPTCProuter.

You can increase sysctl tcp_keepalive_time on both VPS and OpenMPTCProuter or use SSH ServerAliveInterval setting.

Scaff31, 2018-06-19:
every minute my ssh connection falls with tcp_keepalive_time = 36000 how can I fix this?

Ysurac, 2018-06-20:
No problem when tested with latest OMR version. You can also set SSH ServerAliveInterval setting on your client.

closed 2018-08-29 by Ysurac