Ysurac/openmptcprouter #29: Support more arch

Ysurac, 2018-03-30:
For now only Raspberry PI 3 and x86_64 is supported.
To be supported an arch need to work on a 4.14 kernel.

RPI3 is the lowest CPU power I can really support for 100Mb/s.

xslite, 2018-03-31:
Hello, I would love to get an image built of OpenMPTCPRouter ready to deploy for the Rock64 platform
by Pine64 :  https://www.pine64.org/?product=rock64-media-board-computer 

This great little dev board with Rk3328 Arm64 CPU (4x 1.5Ghz), with native Gbit ETH and USB3.0, is more powerfull then PI3+, and cost lower (from USD25 with 1GB Ram).

Already available Linux builds :  https://github.com/ayufan-rock64/linux-build

Thanks a lot for a futur support ;-)

best regards,

XSLite

Ysurac, 2018-03-31:
I need a 4.14 patched kernel, and I need the board to test. I will first test with a Raspberry PI 3 B+, then I can test with this arch when I will have some money ;)

lars18th, 2018-04-03:
Hi,

I suggest to first add support for Docker containers. Then, use it for adding support for different architecture platforms (ARM, Mips, etc.). It will be then more easy for adapt to specific hardware.

Regards.

Ysurac, 2018-04-03:
There is no real problems to run OpenMPTCProuter in a docker container, it's OpenWrt so the Dockerfile for it will work also here.

Docker will not really help to support new architectures. When there is a 4.14 patched to support the arch it's not a problem to add it to OpenMPTCProuter, but to test it really boards are needed.
All arch supported by OpenWrt should work without problems with OpenMPTCProuter. Better if supported with 4.14 kernel.

lars18th, 2018-04-03:
Hi,

The idea is to support Docker images out-the-box. Then enable multiple architectures using QEMU & Docker for each OpenWRT platform.

See the idea here:
https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/

And here a simple script for running it:
https://github.com/justincormack/cross-docker

I hope it helps!

Ysurac, 2018-04-03:
The problem here is not to support ARM, MIPS,... (it's already supported) but to support some specifics board with ethernet drivers, boot procedures,... test it and create images.

lars18th, 2018-04-03:
OK, in this case forget my suggestion!

Ysurac, 2018-04-11:
Raspberry PI 3 B+ support added in 0.15.1

Ysurac, 2018-08-13:
Can be cool to add support for Banana Pi BPI-R2 ( http://wiki.banana-pi.org/Banana_Pi_BPI-R2 )

universat, 2019-01-10:
Bonjour Ysurac, pouvez-vous s'il vous plait ajouter le support pour le modèle WRT1900ACS de Linksys ? il est basé sur le meme CPU que le WRT3200ACM (Armada 385).

Spécifications techniques du WRT1900acs:

CPU1: Marvell 88F6820 (1.6 GHz, 2 cores)
FLA1: 128 MiB (Spansion S34ML01G100TFI00)
RAM1: 512 MiB (SK hynix H5TQ2G63FFR-PBC × 2)

WI1 chip1: Marvell 88W8864
WI1 802dot11 protocols: an+ac
WI1 MIMO config: 4x4:3

WI2 chip1: Marvell 88W8864
WI2 802dot11 protocols: bgn
WI2 MIMO config: 4x4:3

jayanta525, 2019-04-29:
Allwinner based boards are supported. Tested with 100Mbps throughput. Available for purchase on Amazon and Aliexpress ranging from 10$+. Boards in that price range with gigabit ethernet is also available. All of them has OpenWrt support.

OrangePi R1( $14+shipping ): 

- Link: http://www.orangepi.org/OrangePiR1/
- Quad-core Cortex-A7
- 256MB DDR3
- 2x100M Ethernet

OrangePi Zero( $9+shipping ):

- Link: http://www.orangepi.org/orangepizero/
- Quad-core Cortex-A7
- 256/512MB DDR3
- 1x100M Ethernet
- PoE

OrangePi Zero Plus( $15+shipping ):

- Link: http://www.orangepi.org/OrangePiZeroPlus/
- Quad core 64bit Cortex-A53
- 512MB DDR3
- 1 x Gigabit Ethernet

Ysurac, 2019-04-29:
wifi is working on them ?

jayanta525, 2019-04-30:
All the above mentioned boards uses either Realtek RTL8189FTV or XR819 wireless. Wireless works with some modifications at `mac80211.sh` and requires drivers to be installed during compilation. In-short, buggy wireless. And no OpenWrt support by default.

An cheap router like the WR841N( $19 ) repurposed as an access-point with OpenWrt will work better. Also the benefits of VLAN and multiple access-point networks.

This would be the cheapest setup of OpenMPTCProuter for users satisfied with 100Mbps bandwidth.

netkachev, 2020-01-05:
@Ysurac i wonder if you plan to update Docker images at https://hub.docker.com/u/ysurac with latest versions (v0.53.3 at time of writing). generally speaking, some simple ci/cd pipeline for automatically packaging new releases into container images would be handy here i think