Ysurac/openmptcprouter #32: Proposer une documentation simple pour faire le 1er paramétrage

xslite, 2018-03-31:
Bonjour,

ayant assisté une personne voulant paramétrer pour la 1ère fois OpenMPTCPRouter sur un matériel PC, j'ai vu qu'il n'était pas si évident de suivre les étapes nécessaires.

Voici une proposition de tutoriel très simplifié.

Au plaisir de voir s'agrandir la communauté ;-)

XSLite

xslite, 2018-03-31:
See in attached files : PDF and DOCX files

lars18th, 2018-04-03:
Hi,

I suggest to move these guides to the Wiki section. And translate them to different languages.

You agree?

Ysurac, 2018-04-03:
While OpenMPTCProuter 1.0 is not out, I will not translate them. Wizard can (and will) change a lot.

GGFR-Shawn, 2018-06-01:
super explication en tout cas ca m'aurais bien aidé si j'avais trouvé avant j'aurais moins cherché

mykelinho, 2018-09-02:
Bonjour, je n'arrive pas à me connecter à l'interface web de openmptcprouter installer sur mon rpi3.
J'ai reussi une premiere connexion et suite à des mauvais reglages j'ai perdu la connexion.
J'ai donc reformatter ma carte SD pour réinstaller openmptcprouter mais je n'arrive toujours pas a ma reconnecter à l'interface web (adresse ip du pc en 192.168.100.11 et gateway en 192.168.100.1)
Je ne trouve pas de solution...

closed 2020-09-14 by github-actions[bot]