Ysurac/openmptcprouter #41: quelques soucis de résolution DNS

Cmoi20, 2018-04-05:
## Expected Behavior
un accès internet totalement opérationnel via mptcp

## Actual Behavior
avec les paramètres par défaut, certaines applications mobiles ne sont pas opérationnels (ex clash of clan) et certains accès internet ne fonctionnent pas (ex : battlenet ne fonctionne pas bien), ou encore la détection de connexion internet de windows est KO.

## Steps to solve the Problem

  1. dans réseau -> interfaces -> "editer" lan
  2. en bas, Serveur DHCP -> paramètres avancés
  3. ajouter 6,8.8.8.8,8.8.4.4 ou tout autre serveur DNS de votre choix
  4. Enjoy

## Specifications

  - OpenMPTCProuter version: 0.14-6
  - OpenMPTCProuter VPS version: 0.15
  - Platform: windows 10, IOS

Ysurac, 2018-04-05:
glorytun est lancé et fonctionnel ?
Si oui, en mettant dans **Services->Shadowsocks-libev->redir-rules**, *ss-redir for UDP* à *unset* et en sauvant et appliquant ?

Cmoi20, 2018-04-08:
effectivement ca semble mieux avec cette configuration. glorytun est bien lancé et opérationnel.
je laisse tourner quelques jours pour confirmer avant de fermer l'issue.

Imprimerienationale, 2018-04-24:
Alors, après quelques temps, je constate que cela fonctionne quand le service vient d'être activé mais après 2 - 3 jours, cela ne fonctionne plus. je vais tenter de récupérer des infos dans les logs mais ce ne sera pas avant la semaine prochaine.

closed 2018-05-26 by Ysurac