Ysurac/openmptcprouter #43: Redirection de port non fonctionnelle

Adrien54-FR, 2018-04-05:
Bonjour,

Je suis passé à la dernière version d'OpenMPTCP sur le VPS et sur le routeur.
Malgrès cela impossible de faire fonctionner la redirection de port, alors qu'avec des versions 
précédentes j'avais réussi à la faire fonctionner simplement et rapidement.

J'ai installé un TCPDUMP sur le routeur OpenMPTCP :
`root@OpenMPTCProuter:~# tcpdump -i any port 444
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
23:04:06.457364 IP 10.0.0.1.5863 > 10.0.0.2.444: Flags [S], seq 303785297, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:04:06.457504 IP 10.0.0.2.444 > 10.0.0.1.5863: Flags [R.], seq 0, ack 303785298, win 0, length 0
23:04:07.042795 IP 10.0.0.1.5863 > 10.0.0.2.444: Flags [S], seq 303785297, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:04:07.042923 IP 10.0.0.2.444 > 10.0.0.1.5863: Flags [R.], seq 0, ack 1, win 0, length 0
23:04:07.626968 IP 10.0.0.1.5863 > 10.0.0.2.444: Flags [S], seq 303785297, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
23:04:07.627050 IP 10.0.0.2.444 > 10.0.0.1.5863: Flags [R.], seq 0, ack 1, win 0, length 0`

Les paquets arrivent bien jusqu'au routeur mais ils ne vont pas plus loin et ne sont pas transféré sur mon PC... 
La redirection est bien en place dans le routeur ..

[image]

Est-ce-qu'il y a eu une modification dans GloryTun ? 
Comment diagnostiquer le problème ?

## Specifications

  - OpenMPTCProuter version  v0.14-6
  - OpenMPTCProuter VPS version: 0.15

Merci d'avance,

Cordialement

Ysurac, 2018-04-06:
Aucun changement sur Glorytun TCP. Et aucun changement à ce niveau.

Dans la redirection il y a *From any host in wan*, ça devrait être *From any host in vpn*.

Adrien54-FR, 2018-04-06:
Ha oui effectivement je sais pas pourquoi j'ai changé, ça fonctionne mieux comme ça et c'est d'ailleurs plutôt logique...

Sinon ça marche vraiment au top !

Merci :)