Ysurac/openmptcprouter #48: Kernel patch for 4.9 is broken

BigNerd95, 2018-04-27:
When compiling for a target using kernel 4.9, your kernel patch can't be applied.  
This is because openwrt now bumped to 4.9.96 and your patch is for an earlier version.  
You should clone a fixed (older) commit of openwrt to make it build successful.  

I'm trying to find different ways to solve this problem automatically for the future.  
(Like creating the patch dynamically based on the actual kernel version of openwrt).  
I would like to talk about this with you as I'm very interested in porting mptcp on openwrt.  

#### Steps to Reproduce the Problem

  1. Comment line `make IGNORE_ERRORS=m "$@"` in `build.sh`
  2. ./build.sh
  3. make menuconfig
  4. Target System (Atheros AR7xxx/AR9xxx)
  5. make target/linux/compile

Can you list your step to create [690-mptcp_v0.93_b674162.patch](https://github.com/Ysurac/openmptcprouter/blob/master/root/target/linux/generic/hack-4.9/690-mptcp_v0.93_b674162.patch)?  
Did you create it by diffing mptcp kernel source against mainline kernel source?  
Or against kernel with openwrt patches already applied?  

Congratulations for the initiative! ;-)

Ysurac, 2018-04-27:
I don't support any arch using kernel 4.9 officially, so this patch is not maintained for now. This shouldn't be too difficult to fix it for kernel 4.9.96, but it's MPTCP 0.93 and I had some bugs with it. May be better to backport MPTCP 0.94 or, if possible, to update kernel.

The patch is a diff between mptcp kernel source and mainline kernel source, then it's modified to apply to OpenWrt kernel version.

BigNerd95, 2018-04-27:
Ok thank you!

So if tomorrow openwrt update kernel 4.14 you have to manually fix again the 4.14 patch..

Ysurac, 2018-04-27:
Yes. While I'm on master branch it's what I do when there is a problem. I think I will make a stable and unstable release when OpenWrt 18 will be available.

lloovve, 2018-05-11:
when i build target MT7621  linux4.14.36 ,I got the  same error.   is not it be supported linux 4.14.36 ?

Ysurac, 2018-05-11:
It's not he same issue. This issue is for kernel 4.9.
Also latest kernel used is 4.14.37, and it's working for MT7621 even if config file didn't have MPTCP default settings correctly set.

Ysurac, 2018-05-11:
MPTCP 0.93 for kernel 4.9 updated. This compile. Not tested. No support.

closed 2018-05-11 by Ysurac