Ysurac/openmptcprouter #50: Make a DMZ

dougalito, 2018-05-03:
So I would like to make a DMZ on my MPTCP.

I ever tried it when i used OTB official and try to do the same but we have more option here.

So to do my DMZ I only put (in network/firewall/forwards/)

Label : DMZ

Nom : DMZ
Protocole : UDP + TCP
 Zone externe : Tried WAN and VPN
 Port externe : nothing
 Zone interne : LAN
 Adresse IP interne : 192.168.100.30 (my server)
 Port interne : nothing

but my server from outside still not accessible.

Ping on my Public IP and my DNS are ok with ping

Server accessible in lan

Have you got an idea to configure correctly it.

Thanks

## Specifications

 Raspberry PI 3/3 B+: openmptcprouter-v0.16-r6737-99d9c98748-brcm2708-bcm2710-rpi-3-ext4-sdcard.img.gz (2018/04/20 - 18.01Mo)

Ysurac, 2018-05-03:
"*Zone externe*" should always be VPN, data from the server to OpenMPTCProuter are transfered over the VPN.
On the server, did you modify ***/etc/shorewall/rules*** ? You need to uncomment last rules to forward ports from the server to OpenMPTCProuter, this is disabled by default.

dougalito, 2018-05-03:
Ok thanks its working now with dat 

On the server, did you modify /etc/shorewall/rules ? You need to uncomment last rules to forward ports from the server to OpenMPTCProuter, this is disabled by default

closed 2018-05-03 by dougalito