Ysurac/openmptcprouter #52: VnStat Graphs

nicoz76, 2018-05-09:
Nothing display on VnStat Graph

[image]

## Steps to Reproduce the Problem

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.16 r6737-99d9c98748
  - OpenMPTCProuter VPS version: 0.16
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2

thanks

Ysurac, 2018-05-10:
Fixed in latest commit.

closed 2018-05-10 by Ysurac

kevin39, 2019-10-05:
This problem seems back in last version (clean install). Do you confirm ?

Ysurac, 2019-10-05:
Yes, after the update of vnstat graph are not visible. This will be fixed in next release.

max4186, 2019-11-22:
@Ysurac Do you have an ETA for next release? Still broken.