Ysurac/openmptcprouter #59: can not connect router  on MT7621

lloovve, 2018-05-21:
when i try to login openmptcprouter (i build mt7621) , connect 192.168.100.1 but  get nothing
i try login in by ttl usart    use ifconfig get the result:

eth0      Link encap:Ethernet  HWaddr D4:5F:25:EB:31:C1
          inet6 addr: fe80::d65f:25ff:feeb:31c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:62 errors:0 dropped:0 overruns:0 frame:0
          TX packets:66 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6890 (6.7 KiB)  TX bytes:15525 (15.1 KiB)
          Interrupt:20

eth0.1    Link encap:Ethernet  HWaddr D4:5F:25:EB:31:C1
          inet6 addr: fe80::d65f:25ff:feeb:31c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:62 errors:0 dropped:4 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5774 (5.6 KiB)  TX bytes:656 (656.0 B)

eth0.2    Link encap:Ethernet  HWaddr D4:5F:25:EB:31:C1
          inet6 addr: fe80::d65f:25ff:feeb:31c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:46 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:12012 (11.7 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:733 errors:0 dropped:0 overruns:0 frame:0
          TX packets:733 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:45573 (44.5 KiB)  TX bytes:45573 (44.5 KiB)

wan       Link encap:Ethernet  HWaddr D4:5F:25:EB:31:C2
          inet6 addr: fe80::d65f:25ff:feeb:31c2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:37 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:11046 (10.7 KiB)

**is there any error ?**

lloovve, 2018-05-21:
i change /etc/config/network  to
......
config interface 'lan'
        option type 'bridge'
        option ifname 'eth0'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option multipath 'off'
        option ip4table 'lan'

....

and  reboot
it can be login  by lan

Ysurac, 2018-05-21:
I only support supported arch. Without the platform I can't support, test and fix.

closed 2018-05-21 by Ysurac