Ysurac/openmptcprouter #61: after vps install ,ssh can not connect

lloovve, 2018-05-23:
vps  Debian 9.2 64bit  

after intall  shell and  reboot,then vps can not be connetct

if   uninstall shorewall shorewall6  then vps ssh can be connetct

lloovve, 2018-05-23:
yes ,i can not connetct Prot 65222

Ysurac, 2018-05-23:
So when you try with shorewall activated, you can't connect to SSH port 65222, but with shorewall disabled you can connect to SSH port 65222 ?

What is the "net" interface used in /etc/shorewall/interfaces and what is your default interface (check with ```ip -4 route show to default```) ?

lloovve, 2018-05-23:
run shorewall shorewall6
[root /]# ip -4 route
default via 74.120.168.1 dev eth0 
10.0.0.0/24 dev gt-tun0 proto kernel scope link src 10.0.0.1 
10.0.1.0/24 dev gt-tun0 proto kernel scope link src 10.0.1.1 
74.120.168.0/21 dev eth0 proto kernel scope link src 74.120.171.202 

cat /etc/shorewall/interfaces

net	eth0            dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn	gt-tun0        nosmurfs,routefilter,logmartians,tcpflags
////////////////////////////////////////////////////////////////////////////////////////////////////////////

stop shorewall shorewall6   

[root /]# ip -4 route
default via 74.120.168.1 dev eth0 
10.0.0.0/24 dev gt-tun0 proto kernel scope link src 10.0.0.1 
10.0.1.0/24 dev gt-tun0 proto kernel scope link src 10.0.1.1 
74.120.168.0/21 dev eth0 proto kernel scope link src 74.120.171.202 

cat /etc/shorewall/interfaces

#ZONE   INTERFACE       OPTIONS
net	eth0            dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn	gt-tun0        nosmurfs,routefilter,logmartians,tcpflags

lloovve, 2018-05-26:
my  vps is aliyun, the Security Groups  stop  my  connect,  So setting security groups  open 65222  65101 65001 when use aliyun

closed 2018-05-26 by lloovve