Ysurac/openmptcprouter #63: option/documentation for udp through glorytun vpn

dur3x, 2018-05-24:
## Expected Behavior

I'm trying to force all my traffic (e.g. QUIC protocol https://ma.ttias.be/googles-quic-protocol-moving-web-tcp-udp/) to go through the glorytun vpn so that I can use all the bandwith by using for example youtube with QUIC

## Actual Behavior
All the UDP traffic goes through the shadowsocks Tproxy (this is done only through one wan interface(the default one set as master), not multiple)

## Steps to Reproduce the Problem

  1. Use chrome (with the default option so that quic protocol is enabled) and watch video on youtube.
  2. Observ that all traffic goes on only one interface (my goal is to use all ones)

## Specifications

  - OpenMPTCProuter version: latest
  - OpenMPTCProuter VPS version: latest
  - OpenMPTCProuter platform: (RPI3/x86_64) all

Ysurac, 2018-05-24:
You can disable UDP redirection though ShadowSocks in Services->ShadowSocks->Redir Rules and set ss-redir for UDP to unset. This will use Glorytun VPN if enabled, else you Master WAN connection.

dur3x, 2018-05-24:
Thanks for your reply, I will try that again this evening but I think that
when I checked my settings this parameter was already shown as TCP unset &
UDP unset.
I keep you in touch.

Ysurac, 2018-05-24:
True, there is a bug in shadowsocks interface since a change in LuCi.
This will be fixed in next release.

dur3x, 2018-05-24:
Ok so I just tried by modifying my configuration in  /etc/config/shadowsocks-libev
like that: (remove udp redirection + restart)
https://github.com/Ysurac/openmptcprouter-feeds/commit/9028cc1823715c008fbba60e1e3d8754d70b57c8

but it looks this is not sufficient, for example if I'm trying that it's not working:
`nslookup google.fr 8.8.8.8`

Ysurac, 2018-05-24:
I forget to create a rules to allow forward from LAN to VPN in firewall...
This will be fixed in next release that should be available in one or two hours.
Thanks.

dur3x, 2018-05-24:
Thanks a lot ! It's ok with your latest commit https://github.com/Ysurac/openmptcprouter-feeds/blob/75c70fa721bbc4772de231b8eb6a3558302a2900/luci-app-glorytun/root/etc/uci-defaults/40_luci-glorytun

closed 2018-05-26 by Ysurac