Ysurac/openmptcprouter #64: Use Glorytun VPN for UDP instead of shadowsocks

Ysurac, 2018-05-24:
There is some problems with shadowsocks UDP.
Disable UDP shadowsocks by default.
Maybe detect use shadowsocks UDP only when Glorytun VPN is down.

Ysurac, 2018-05-25:
Since release 0.20, UDP doesn't use shadowsocks by default.
Need to make a script to use it if glorytun is down (at user choice).

Ysurac, 2018-06-01:
This will be available in next release.

closed 2018-06-20 by Ysurac

Bumpalert, 2021-06-02:
@Ysurac When I need port forwarding & UDP video uploading, which VPN i should use. Disabling Glorytun TCP, i am lossing port forwarding. As shadowsocks UDP not working, if I enable Glorytun UDP & Glorytun TCP together which will be choosen by live streaming video content?

Ysurac, 2021-06-02:
Port forwarding work with all VPNs available in wizard page. You can't enable multiples VPN at the same time.

Bumpalert, 2021-06-02:
@ysurac I noticed that If I change to vray & glorytun udp from shadow socks & glorytun tcp, the port forwarding will stop working and vpn wont connect on 0.58RC2. I tried 2-3 times to make sure that this is the problem and undoing the change and setting tcp vpn and port forwarding will start working.

[image]
[image]

 I didn't upgrade after 0.58rc2 as server script update from 1025 may need to reset vps,as running 1024 to 1025 didnt upgrade without reseting the vps. Is there any  easy way to uninstall existing vps script and install updated one.

Ysurac, 2021-06-02:
You must upgrade to 0.1025 on VPS side if you want 0.58 to work.

Bumpalert, 2021-06-02:
> 
> 
> You must upgrade to 0.1025 on VPS side if you want 0.58 to work.

I am actually on 0.1025. I didn't upgrade after that because I don't wanted to reset vps frequently(if any new vps script 0.1025+ is released with 0.58rc5) as 0.58rc2 works stable for me. 0.58rc2 port forwarding is working on tcp vpn and shadow socks. But as you suggested earlier, when I moved to glorytun UDP & VRAY on wizard, port forwarding stopped working & vpn disconnected(I did the setting change actually on the omr forwarded port on vps through internet ) . If I put back to tcp glorytun(using local network as no port forwarding active) port forwarding will work as earlier.