Ysurac/openmptcprouter #70: Test MLVPN over MPTCP

Ysurac, 2018-06-01:
Using https://github.com/markfoodyburton/MLVPN/tree/new-reorder
Need to make a package and LuCI interface if it's working well.

aredan, 2018-06-01:
I was thinking of using MLVPN but read a lot about it and it has a few issues, there is not much activity on the GitHub issues page. 

What are you planning to replace with MLVPN?
Will it be an interchangeable option with Glorytun for example?

I'm looking for a way to use three internet connections (2 x Fiber + 1backup 3G USB) changing the less of internal routing of my network.

Ysurac, 2018-06-01:
I will make some tests/benchmarks, and if it's usable, this will be interchangeable with Glorytun TCP/UDP.
It's only a test for now.

Ysurac, 2018-06-11:
MLVPN support is added in latest release. Mark Foody Burton version crash on RPI, will try to investigate that later.

aredan, 2018-06-11:
I will update to this version to test MLVPN, I'm running on x86_64.

nileio, 2018-07-01:
can someone report results of using MLVPN instead of glorytun ?

Ysurac, 2018-07-03:
Master tree use Markfoodyburton MLVPN version. It's working better than previous version but it's still slower than ShadowSocks with MPTCP.

beyondhong, 2019-04-04:
Hi @Ysurac ,
I tested the MLVPN, using two 4G modules, but it's not working good. Link aggregation did not take effect. It's even worse than single link. I opened an issue here:
https://github.com/zehome/MLVPN/issues/121
Have you got any suggestion? Thanks!

Ysurac, 2019-04-04:
I'm using https://github.com/markfoodyburton/MLVPN/tree/new-reorder fork in OpenMPTCProuter, but as I said I don't find the result really good.

beyondhong, 2019-04-04:
> I'm using https://github.com/markfoodyburton/MLVPN/tree/new-reorder fork in OpenMPTCProuter, but as I said I don't find the result really good.

Yeah I'm using that version too. Now I'm trying to test Glorytun, so how will I start with it?
I am testing on a mtk(mt7620) router running dd-wrt.

closed 2019-12-20 by Ysurac

Neustradamus, 2019-12-20:
@Ysurac: It is solved?

Ysurac, 2019-12-20:
Yes it's solved: MLVPN is no more used.
Too many crash, not usable anymore.