Ysurac/openmptcprouter #71: routing using OpenMPTCProuter.

aredan, 2018-06-01:
## Expected Behavior

 Have an x.x.x.x/24 that I want to route via the VPS, public IP are being announced to the Internet using the VPS.

## Actual Behavior

Not tested yet.

## Steps to Reproduce the Problem

Not tested yet.

## Specifications

  - OpenMPTCProuter version: latest
  - OpenMPTCProuter VPS version: latest
  - OpenMPTCProuter platform: x86_64

 I want to use OpenMPTCProuter as a WAN aggregator but also I want to route a Public block of IP address back to my DMZ, right now I'm doing it using TincVPN.

This can be done on top of Glorytun and use benefits of MPTCP?

Ysurac, 2018-06-01:
Yes, this should be possible. Glorytun is a VPN. You will have to modify the configuration of shorewall on the VPS (if you use the install script), but else this should be the same network config as TincVPN.

closed 2018-07-07 by Ysurac

nileio, 2018-08-23:
sorry to dig this up again @Ysurac .. do you mean it is possible to replace Glorytun with TincVPN ? or rather you say that we can run TincVPN on top of Glorytun ? i have a similar use case and wonder if i can use TincVPN to do that .. thanks

Ysurac, 2018-08-23:
Both are possible, but this would be more TincVPN over ShadowSocks, in TCPonly mode it can use MPTCP.