Ysurac/openmptcprouter #72: Add possibility to change IP of openmptcprouter

constant1010, 2018-06-02:
Is possible to add possibility to change IP of router?

It would be practical to have file in .txt on /boot partition with adress ip of router to set, and I just thought why not shadowsocks key ,glorytun key, wan adress, etc.

Ysurac, 2018-06-02:
You can change IP in **Network->Interfaces**.
I can't really see a case why it would be practical to have a file in /boot. All configuration is in **/etc/config/**.

constant1010, 2018-06-11:
It would be practical, when you just flash sd card, you don't must to change IP of device to set up IP of router.
There is IP of **Port Redirection** to change, is there other parameter?

Ysurac, 2018-06-11:
You can have multiple IP on an interface, if you really don't want that you can edit /etc/config/network on the sd card.
In next release no parameters will need to be changed after IP change.

closed 2018-06-25 by Ysurac