Ysurac/openmptcprouter #74: Update OpenMPTCProuter VPS script

teraska47, 2018-06-04:
Hello, 
I just saw that the VPS script has been updated, what is the update procedure ?

Erik

Ysurac, 2018-06-04:
You only have to run it again. Latest version will keep shadowsocks and glorytun keys, it also keep shorewall rules.

teraska47, 2018-06-04:
Ok thanks Ysurac, that's what I wanted to know

thomasforte, 2018-06-05:
If i run the latest version of the script (commit e6bdcb066f4836ca5c9c861a17638e8dffb39097), i don't have the summary of the configuration at the end of the script (and no "/root/openmptcprouter_config.txt" generated).
Instead, i have : "OpenMPTCP router VPS is now updated ..."

Ysurac, 2018-06-05:
Strange. Latest version test if shadowsocks-libev config exist. I've updated to check in /etc/motd

thomasforte, 2018-06-05:
It's ok now (with commit 36a727d7d55fbcc6a0d45808d48ace4cb137a8ee). The first run of the script generate the /root/openmptcprouter_config.txt.

thomasforte, 2018-06-05:
So, a second run of the script give the summary. There are some "OpenMPCTProuter" instead of "OpenMPTCProuter" (lines 210 à 215 in the script).

thomasforte, 2018-06-06:
Ok, it works well now.
Just a problem if i choose to not recompile the kernel again (see below) : 

`Ign:1 http://ftp.fr.debian.org/debian stretch InRelease
Atteint:2 http://security.debian.org/debian-security stretch/updates InRelease                                        
Atteint:3 http://deb.debian.org/debian stretch-backports InRelease                                                    
Atteint:4 http://ftp.fr.debian.org/debian stretch-updates InRelease                                                   
Atteint:5 http://ftp.fr.debian.org/debian stretch Release                    
Lecture des listes de paquets... Fait                                        
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
dirmngr is already the newest version (2.1.18-8~deb9u1).
patch is already the newest version (2.7.5-1+b2).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 1 non mis à jour.
Ign:1 http://ftp.fr.debian.org/debian stretch InRelease
Atteint:2 http://security.debian.org/debian-security stretch/updates InRelease
Atteint:3 http://deb.debian.org/debian stretch-backports InRelease    
Atteint:4 http://ftp.fr.debian.org/debian stretch-updates InRelease   
Atteint:5 http://ftp.fr.debian.org/debian stretch Release
Lecture des listes de paquets... Fait
--2018-06-06 09:33:16--  http://www.openmptcprouter.com/kernel/linux-image-4.14.24-mptcp-64056fa.amd64.deb
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 37536054 (36M) [application/octet-stream]
Sauvegarde en : « /tmp/linux-image-4.14.24-mptcp-64056fa.amd64.deb »

/tmp/linux-image-4.14.24-mptcp-64056fa.amd 100%[=====================================================================================>]  35,80M  17,6MB/s    in 2,0s    

2018-06-06 09:33:18 (17,6 MB/s) — « /tmp/linux-image-4.14.24-mptcp-64056fa.amd64.deb » sauvegardé [37536054/37536054]

--2018-06-06 09:33:19--  http://www.openmptcprouter.com/kernel/linux-headers-4.14.24-mptcp-64056fa.amd64.deb
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 9124188 (8,7M) [application/octet-stream]
Sauvegarde en : « /tmp/linux-headers-4.14.24-mptcp-64056fa.amd64.deb »

/tmp/linux-headers-4.14.24-mptcp-64056fa.a 100%[=====================================================================================>]   8,70M  17,5MB/s    in 0,5s    

2018-06-06 09:33:19 (17,5 MB/s) — « /tmp/linux-headers-4.14.24-mptcp-64056fa.amd64.deb » sauvegardé [9124188/9124188]

Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
rename is already the newest version (0.20-4).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 1 non mis à jour.
(Lecture de la base de données... 59431 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../linux-image-4.14.24-mptcp-64056fa.amd64.deb ...
Ok, aborting, since modules for this image already exist.
dpkg: erreur de traitement de l'archive /tmp/linux-image-4.14.24-mptcp-64056fa.amd64.deb (--install) :
 le sous-processus nouveau script pre-installation a retourné une erreur de sortie d'état 1
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.14.24-mptcp-64056fa+ /boot/vmlinuz-4.14.24-mptcp-64056fa+
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 4.14.24-mptcp-64056fa+ /boot/vmlinuz-4.14.24-mptcp-64056fa+
Des erreurs ont été rencontrées pendant l'exécution :
 /tmp/linux-image-4.14.24-mptcp-64056fa.amd64.deb
`

Ysurac, 2018-06-06:
I will check how I can force replace.

Ysurac, 2018-06-11:
Done in latest OpenMPTCProuter VPS script

closed 2018-06-11 by Ysurac