Ysurac/openmptcprouter #75: Changing the Glorytun gt-tun0 IP address on VPS and OMR.

aredan, 2018-06-06:
## Expected Behavior

 I cant disable the DHCP of one of the WAN modem, others devices need this DHCP. the subnet is 10.0.0.0/24.

## Actual Behavior

ICMP and UDP aren't working, I think that the Glorytun IP on the server and OMR are in conflict with the IP of WAN2 (10.0.0.254).

## Steps to Reproduce the Problem

  1.  put any WAN subnet in 10.0.0.0/24
  2.  I think it will conflict with Glorytun

## Specifications

  - OpenMPTCProuter version: 0.21
  - OpenMPTCProuter VPS version: 0.20
  - OpenMPTCProuter platform: x86_64

I took a look in the VPS and can see:

```
gt-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet 10.0.0.1  netmask 255.255.255.0  destination 10.0.0.1
        inet6 fe80::3bcb:2f07:6a48:3c1e  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 1773303  bytes 590768566 (563.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2629056  bytes 1409681633 (1.3 GiB)
        TX errors 0  dropped 7017 overruns 0  carrier 0  collisions 0
```

 How can I change this subnet to something more harder to conflict with? I mean, set up the gt-tun0 and others Glorytun interface to  10.255.250.0/24 and increment from that instead of 10.0.0.0/24 and 10.0.1.0/24

The device used for OMR has multiple ethernet ports, so I'm receiving DHCP announce in the WAN2 which is the one with 10.0.0.0/24.

Right now I don't have access to my OMR, will update the post with images later.

Thank you!

Ysurac, 2018-06-06:
IP for Glorytun TCP VPN come from the DHCP server defined in ```/lib/systemd/network/glorytun-tcp.network```

aredan, 2018-06-06:
I did the changes, the new IP address was reflected in the Interface page of OMR but from CLI when I did an ifconfig I can see that tun0 was still using 10.0.0.2 as the assigned IP address.  I did a reboot to my OMR but it didn't change. Also, change the Glorytun configuration and set both localip and remoteip.

I'll wait for a new release with this changes.

Thanks!

Ysurac, 2018-06-11:
Available in latest OpenMPTCProuter VPS script and latest OpenMPTCProuter release

closed 2018-06-11 by Ysurac