Ysurac/openmptcprouter #76: [question] how should my devices connected to Raspberry Pi 3 after installing precompiled image?

c1ngular, 2018-06-07:
amazing work , i am trying to install this on my Raspberry Pi 3 b+, but i can not figure out after installation , how should my other devices(mobile,PC) connect to Pi (in this case Pi is a router, am i right ) ?  is it possible if i just want MPTCP installed on my Pi without additional softwares , so i my Pi could communicate with my MPTCP-enabled server ? would you be so kind to tell me how to solely install MPTCP on Pi 3 B+ ?

Thank you

aredan, 2018-06-07:
I don't know how to only install an enable MPTCP kernel into the Pi, with this image (OMR) the connections for the Pi are like the one for OverTheBox, you disable DHCP on all your ISP modems and connect them into a switch (or used the ports of one modem), after that you configure the WAN interfaces on the OMR for each modem and use the 192.168.100.1 ( if you didn't change it) as your gateway.

I did try to do something like OMR on Debian, I followed the step by step in the MPTCP website but at the end the is easier to just install this platform.

In my case, I used a machine with 6 ethernet ports to avoid conflicts and change the minimum on my network.

c1ngular, 2018-06-07:
@aredan thank you , i am using my Pi 3b/3b+ to stream local video to remote server, i have streaming software installed on my Pi already , but sometimes internet connection to Pi is slow/unstable, so i want to utilize MPTCP to aggregate Wifi and ethernet bandwidth to secure my streaming service, my server is already MPTCP-enabled . 

i don't really understand why we need those additional softwares besides the MPTCP, i get the shadowsocks part as well , not the rest though . 

or should i buy a compatible hardware router , and try this out ? 
what am i missing here ?

thanks again !

Ysurac, 2018-06-07:
OpenMPTCProuter is based on OpenWrt Linux distribution. It's a router, so there is every software needed for a router. You can find software added to OpenWrt here: https://github.com/Ysurac/openmptcprouter-feeds

If you need only MPTCP, follow doc on http://multipath-tcp.org

closed 2018-07-07 by Ysurac