Ysurac/openmptcprouter #78: Very Unstable connexion

xslite, 2018-06-08:
## Expected Behavior

the connexion with +100Mbs cable modem + 4G modem (up to 100Mbs) should be stable and fluent...

## Actual Behavior

the bandwidth is extremely varying (50 to 150 Mbs with the current setup), and not stable at all. The MPTCP graphs shows a lot of pikes. (check screenshot in attached file)

When navigating on various website, initial connexion is very slow, and issues a lot of timeouts. Needs to restart page several times before getting the normal full design.

A lot of SSL website cannot be seen at all. 

## Steps to Reproduce the Problem

VPS is Online.net dedibox with 1Gbs uplink... never shown any problem.

  1. launch a big download
  2.check MPTCP graphs
[image]

  3.

## Specifications

  - OpenMPTCProuter version: 0.21
  - OpenMPTCProuter VPS version: 0.19 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2018-06-08:
Can you give an example of website you can't see ?
IPv6 is activated or not ? In 0.21 there is a bug with IPv6 tunnel that will be solved in next version.

abilly007, 2018-06-29:
Did you not find this problem, in this MPTCP flow chart, the average flow rate of the wan1 interface is 120.08Mbps, the light blue color in the diagram, and the average flow rate of the wan2 interface is 1.13Mbps, in the diagram the light green color block, the obvious wan2 flow is far smaller than the wan1 flow, but in the diagram it can be seen that green is higher than blue.

closed 2018-08-05 by Ysurac