Ysurac/openmptcprouter #79: Port Forward

plastoc, 2018-06-09:
## Expected Behavior
i  try to redirect port but dont work 

## Actual Behavior
the connection work perfectly , i can access internet through vpn.

i have redirect port in NAT interface on the raspberry pi but dont work
is it possible to view on VPS if connection is realy reveive and where its redirected?
in shorewall/rules ,  port are redirect on 10.0.0.2 but on VPS i try to ping 10.0.0.2 but dont work ,is it normaly?

## Specifications

  - OpenMPTCProuter version: v0.21-3
  - OpenMPTCProuter VPS version:  i dont know
  - OpenMPTCProuter platform: (RPI3/x86_64)

regards
jules

Ysurac, 2018-06-09:
This will be fixed in next release, a service will check what IP is realy available.
Before that IP are available via DHCP, so if MAC address change there is a bug.

Next release should be available this monday (I hope)

Ysurac, 2018-06-11:
Latest OpenMPTCProuter 0.22 release should fix some bugs.
You can follow doc here: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

Kaffeine75, 2018-06-13:
Depuis mon passage en 0.22 (VPS et Router), la redirection de port ne se fait plus alors qu'elle fonctionnait très bien auparavant. J'ai tenté de faire une réinstallation complète mais cela n'a rien changé.

Je vois que dans la doc s'est rajouté une partie SNAT que je n'avais pas eu besoin de configurer lorsque j'avais fait ma configuration il y a quelques mois. Mais cette partie semble bugguer, ce qui m'empêche d'y rajouter une règle...

Ysurac, 2018-06-13:
Le SNAT est nécessaire dans certains cas et je ne vois aucun bug dans cette partie.
La dernière version d'OpenMPTCProuter VPS est la 0.24 et la 0.22.1 pour OpenMPTCProuter.

Kaffeine75, 2018-06-13:
Merci de la réponse rapide.

J'ai bien les dernières versions VPS (24) + Router (j'avais la 22, je viens de passer sur la 22.1).

Pour le bug du SNAT, en faites, cela venait juste que contrairement à l'ajout de port, il fallait remplir tous les champs pour la création de la règle, donc effectivement cette partie fonctionne. Malheureusement l'ajout de SNAT n'a pas résolu mon problème de base.

J'avais précédemment la version VPS 22 + Router 21 et tout fonctionnait parfaitement. J'ai des redirections vers deux serveurs. Pour éliminer que le problème ne venait pas d'ailleurs, j'ai isolé les deux serveurs sur la box de mon premier opérateur et fait la conf NAT dessus et pas de soucis. 

Es-ce que si je repasse le router en 21, ce dernier sera-t-il compatible avec le VPS en 24 ?

Ysurac, 2018-06-13:
La dernière version fonctionne. Il faut bien suivre la documentation, et vérifier que le VPN fonctionne correctement. Attention à avoir la bonne IP dans /etc/shorewall/rules ou d'avoir la variable $OMR_ADDR (la dernière version d'OpenMPTCProuter VPS fait maintenant la modification automatiquement).

plastoc, 2018-06-13:
super merci
effectivement en mettant a jour le vps et le roteur avec la derniere verison , tout fonctionne super.

petite question peut on en fonction du type de donnée ( p2p) rediriger non pas du lan vers le vpn mais vers un wan, en effet j'agreg un adsl + une 4g et souhaite surtout utilise rla 4g pour me connecter de l'exterieur et je ne souhaiterais pas vider mon forfait 4g avec du p2p
du coup peut on choisir la destination en fonction des service , ou priorisé k'upload sur un canal et le download sur l'autre?
merci
jules

Kaffeine75, 2018-06-14:
Sur shorewall, j'ai bien : 
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999

J'ai tenté le downgrade de la partie router en 21 et en 20 mais cela ne semble pas rétrocompatible avec le VPS en 24 puisque je perds la partie Glorytun
.

Pour l'instant, j'ai du coup réactivé le DHCP et le NAT sur la box de la connexion 1 et mis mes serveurs en directes derrières. Solution temporaire qui me rend l'accès de mes serveurs depuis l'extérieur mais sans l'agrégation puisqu'ils ne transitent plus via OpenMPTCPRouter. Et qui me permet tout de même de profiter de l'agrégation via le reste réseau dont les IPs sont fixés dans ma conf OpenMPTCPRouter. 

Je viens de passer deux soirées à tenter de résoudre le problème, j'essayerais de faire d'autres tests dès que le temps me le permettra...

Au passage, suis-je le seul à avoir ce problème ?

Ysurac, 2018-06-14:
Dans params.vpn, OMR_ADDR correspond bien à l'adresse d'OpenMPTCProuter ?
Cette adresse répond au ping depuis le VPS ?

Kaffeine75, 2018-06-14:
Dans params.vpn j'ai : 
OMR_ADDR=10.255.255.2
OMR_IFACE=gt-tun0

Et effectivement, ça ne ping pas contrairement à 10.255.255.1. Le problème était possiblement là. Je tente la bascule de mes serveurs dès que j'ai un moment et te tiens au courant.

A noter que je n'ai pas touché à ces paramètres de mon côté et que j'avais fais une réinstallation depuis 0 mardi soir des parties VPN et Router. Si cette information peut être utile ;).

Ysurac, 2018-06-14:
Le tunnel n'est peut être pas up. 10.255.255.2 doit correspondre à l'IP de OpenMPTCProuter (et elle est normalement détectée automatiquement si le tunnel fonctionne).
Sur OpenMPTCProuter qu'elle est l'IP de l'interface omrvpn ? Et depuis OpenMPTCProuter il est possible de pinger l'adresse du tunnel côté VPS (ici 10.255.255.1) ?

J'ai mis à le script OpenMPTCProuter VPS à jour un certain nombre de fois suite à quelques soucis et un effort de simplification de la configuration de Shorewall. La dernière est la 0.25

Kaffeine75, 2018-06-14:
OMRVPN n'était pas activé... Je l'ai activé en lui définissant comme hôte 10.255.255.2.

J'ai ensuite voulu passer le VPS en 25 mais le script termine sur une erreur 403 : 
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 94.23.252.192
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:80... connected.
HTTP request sent, awaiting response... 403 Forbidden
2018-06-14 14:43:36 ERROR 403: Forbidden.

Après redémarrage de mon VPS et du Router. J'ai rebasculé mes deux serveurs sur l'OpenMPTCPRouter mais le problème est toujours présent. Je vois également une petite évolution du params.vpn : 
VPS_ADDR=10.255.254.1
OMR_ADDR=10.255.254.2
VPS_IFACE=gt-tun0
Je ne sais pas pourquoi il a décidé de passer de 255 à 254... Du coup, es-ce qu'il faut que je passe le paramètre hôte d'OMRVPN en 254 également ?

Pour répondre à tes questions, depuis le Router et le VPS, le ping se fait bien vers 10.255.254.2 et 10.255.254.1.

Par contre Shorewall ne veut plus démarrer : 
root@vps530217:~# systemctl status shorewall.service
● shorewall.service - Shorewall IPv4 firewall
   Loaded: loaded (/lib/systemd/system/shorewall.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-06-14 15:08:26 CEST; 1min 24s ago
  Process: 2846 ExecStart=/sbin/shorewall $OPTIONS start $STARTOPTIONS (code=exited, status=255)
 Main PID: 2846 (code=exited, status=255)

Jun 14 15:08:26 vps530217 shorewall[2846]: Compiling Accept Source Routing...
Jun 14 15:08:26 vps530217 shorewall[2846]: Compiling /etc/shorewall/snat...
Jun 14 15:08:26 vps530217 shorewall[2846]: Compiling MAC Filtration -- Phase 1...
Jun 14 15:08:26 vps530217 shorewall[2846]: Compiling /etc/shorewall/rules...
Jun 14 15:08:26 vps530217 shorewall[2846]:    ERROR: Undefined shell variable ($OMR_ADDR) /etc/shorewall/rules (line 60)
Jun 14 15:08:26 vps530217 root[2963]: ERROR:Shorewall start failed
Jun 14 15:08:26 vps530217 systemd[1]: shorewall.service: Main process exited, code=exited, status=255/n/a
Jun 14 15:08:26 vps530217 systemd[1]: Failed to start Shorewall IPv4 firewall.
Jun 14 15:08:26 vps530217 systemd[1]: shorewall.service: Unit entered failed state.
Jun 14 15:08:26 vps530217 systemd[1]: shorewall.service: Failed with result 'exit-code'.
root@vps530217:~# systemctl restart shorewall
Job for shorewall.service failed because the control process exited with error code.
See "systemctl status shorewall.service" and "journalctl -xe" for details.
root@vps530217:~# systemctl status shorewall.service
● shorewall.service - Shorewall IPv4 firewall
   Loaded: loaded (/lib/systemd/system/shorewall.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-06-14 15:09:56 CEST; 3s ago
  Process: 3275 ExecStart=/sbin/shorewall $OPTIONS start $STARTOPTIONS (code=exited, status=255)
 Main PID: 3275 (code=exited, status=255)

Jun 14 15:09:56 vps530217 shorewall[3275]: Compiling Accept Source Routing...
Jun 14 15:09:56 vps530217 shorewall[3275]: Compiling /etc/shorewall/snat...
Jun 14 15:09:56 vps530217 shorewall[3275]: Compiling MAC Filtration -- Phase 1...
Jun 14 15:09:56 vps530217 shorewall[3275]: Compiling /etc/shorewall/rules...
Jun 14 15:09:56 vps530217 shorewall[3275]:    ERROR: Undefined shell variable ($OMR_ADDR) /etc/shorewall/rules (line 60)
Jun 14 15:09:56 vps530217 root[3392]: ERROR:Shorewall start failed
Jun 14 15:09:56 vps530217 systemd[1]: shorewall.service: Main process exited, code=exited, status=255/n/a
Jun 14 15:09:56 vps530217 systemd[1]: Failed to start Shorewall IPv4 firewall.
Jun 14 15:09:56 vps530217 systemd[1]: shorewall.service: Unit entered failed state.
Jun 14 15:09:56 vps530217 systemd[1]: shorewall.service: Failed with result 'exit-code'.

Ysurac, 2018-06-14:
relance le script sur le VPS, il n'a pas installé tout ce qu'il faut pour le firewall.

Kaffeine75, 2018-06-14:
Merci beaucoup, ça refonctionne :). Je pense que j'avais du tomber sur une version 24 bugger du VPS ^^.
Bon maintenant, plus qu'à attendre la propagation DNS de tous mes noms de domaines...

closed 2018-06-14 by Ysurac