Ysurac/openmptcprouter #80: Request built-in support for USB Wi-Fi adapters (via kmod-rt2800-usb)

reedog117, 2018-06-10:
## Expected Behavior

 Attempted to install kmod-rt2800-usb for USB Wi-Fi support. 

## Actual Behavior

Unable to install because of the kernel dependency. Is there a way around this? My plan is to use multiple hotspots as the different WAN connections. Perhaps the precompiled image can have this support baked in? Or how would I build the image with this support manually?

## Steps to Reproduce the Problem

  1.  ssh or console to openmptcprouter
  2.  opkg update
  3.  opkg install kmod-rt2800-usb

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.21-3-g14c063c-r7093-4fdc6ca31b-brcm2708-bcm2710-rpi-3-ext4-sdcard
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter platform: RPI3 64-bit

Ysurac, 2018-06-10:
This will be added in next release that should be available this Monday.

reedog117, 2018-06-10:
Can we also have libimobiledevice to enable USB iPhone/iPad tethering?

Ysurac, 2018-06-10:
This will also be added in next release.

closed 2018-06-11 by Ysurac