Ysurac/openmptcprouter #81: Update logiciels

nicoz76, 2018-06-12:
## Expected Behavior

Problem when MAJ packets.

## Actual Behavior

## Steps to Reproduce the Problem

[image]

[image]

## Specifications

  - OpenMPTCProuter version: v0.22 r7161-c8677ca89e
  - OpenMPTCProuter VPS version:   last
  - OpenMPTCProuter platform: Raspberry Pi 2 Model B Rev 1.1

Ysurac, 2018-06-12:
I'm not able to reproduce the problem. Maybe a small problem with openwrt servers or a download during openwrt snapshots updates.

nicoz76, 2018-06-12:
after a reboot,  the same : 
[image]

Ysurac, 2018-06-12:
You can try to disable IPv6, in *System->OpenMPTCProuter->advanced settings*

nicoz76, 2018-06-12:
ok ,  k'ai coché : 
[image] 

je teste  :)

nicoz76, 2018-06-13:
c'est pareil, , même après désactivation IPV6 et reboot.

closed 2018-08-05 by Ysurac