Ysurac/openmptcprouter #85: Two openmtcprouter for one vps

constant1010, 2018-06-15:
Is it possible to connect 2 routers, with different location, with one vps server? Maybe it's a good idea to reduce the cost when we have two location.

Scaff31, 2018-06-15:
i think it's possible if your vps've two public IP

Ysurac, 2018-06-15:
It's possible even with one public IP, but it's not supported by the OpenMPTCProuter VPS script, you would have to make many modifications.
But a VPS is not really expensive this would limit the available bandwidth (that is often low for a VPS...)

Scaff31, 2018-06-17:
And if vps are two public ip can you explain how setup vps ?

Ysurac, 2018-06-20:
This would need to have shadowsocks and glorytun listen on each IP and to configure shorewall.
For now this will not be a supported setup.

closed 2018-06-22 by Ysurac

xabix, 2019-09-03:
Hello, I was wondering if we are supposed to be able to upgrade the packages? I can easily to an 'updade' but I can't seem to be able to upgrade any of the packages as it says the filesystem is in read only with 0.51-5. Any idea if we are suppose to be able to upgrade the packages or not?

Ysurac, 2019-09-04:
You shouldn't upgrade packages but filesystem shouldn't be read only.
But this is not related to the main issue...

RealVaVa, 2021-05-10:
> It's possible even with one public IP, but it's not supported by the OpenMPTCProuter VPS script, you would have to make many modifications.
> But a VPS is not really expensive this would limit the available bandwidth (that is often low for a VPS...)

I really need to get quality on few locations (for SIP), the bandwich for each is <1-2Mbps, it should be available for many VPS.
I would like to save compilation of your scheme (openmtcp router and VPS), because it really works (thanks a lot).  But could you share what kind modifications  I need to make for multiple connections (2 and more)?