Ysurac/openmptcprouter #87: OpenMPTCProuter on esxi vm

Scaff31, 2018-06-16:
## Expected Behavior

  Try to setup openMPTCProuter on esxi virtual machine but no idea why no work

## Actual Behavior

SYSLOG:
Sat Jun 16 08:59:03 2018 daemon.err /usr/bin/ss-local[10003]: getpeername: Socket not connected
Sat Jun 16 08:59:03 2018 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 23.96.52.53)
Sat Jun 16 08:59:06 2018 user.notice post-tracking-post-tracking: wan1 switched off
Sat Jun 16 08:59:15 2018 daemon.err uhttpd[4936]: Failed to establish connection
Sat Jun 16 08:59:15 2018 daemon.err /usr/bin/ss-local[10003]: getpeername: Socket not connected
Sat Jun 16 08:59:28 2018 daemon.err /usr/bin/ss-local[10003]: getpeername: Socket not connected

Kernel log:
[    7.375155] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.841431] random: fast init done
[    9.246354] ip_local_port_range: prefer different parity for start/end values.
[   10.303784] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   10.303859] 8021q: adding VLAN 0 to HW filter on device eth0
[   10.306312] device eth0 entered promiscuous mode
[   10.312412] IPv6: ADDRCONF(NETDEV_UP): wan1: link is not ready
[   10.312482] 8021q: adding VLAN 0 to HW filter on device wan1
[   10.323299] IPv6: ADDRCONF(NETDEV_UP): wan2: link is not ready
[   10.323371] 8021q: adding VLAN 0 to HW filter on device wan2
[   10.324062] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   10.325174] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   10.326555] IPv6: ADDRCONF(NETDEV_CHANGE): wan1: link becomes ready
[   10.326670] IPv6: ADDRCONF(NETDEV_CHANGE): wan2: link becomes ready
[   41.509171] random: crng init done
[  136.346754] 8021q: adding VLAN 0 to HW filter on device wan1
[  136.391859] 8021q: adding VLAN 0 to HW filter on device wan2

## Steps to Reproduce the Problem

  1. Install on ESXI VM
  2. Start VM and setup
  3. Look log :)

## Specifications

  - OpenMPTCProuter version: 2.2.4-1
  - OpenMPTCProuter VPS version:  debian 9 stretch
  - OpenMPTCProuter platform: x86_64

Scaff31, 2018-06-16:
Ok I finally managed to put the router on a VM ESXI

For those who wish to do the same I give you the procedure to follow.

1- Download image x64_86

2- With qemu-img, convert the image to vmdk esxi
qemu-img convert -f raw -O vmdk openmptcp-*.img openmptcp.vmdk

3- Create the virtual machine:
Compatibility: 6.0 ( en 6.5 ca does not work)
System Family: Other
OS guest version: other 32bit linux

CPU =1
RAM = 256

delete hard disk, SCSI controller, CD drive

Add a disk that will be the image you converted
Network adapter: e1000

You have completed the configuration of the VM it is not finished
Go to networking then virtual switch select vswith0 and change the settings. In security, activate promiscuity mode.

You're done

Translated with www.DeepL.com/Translator

closed 2018-06-16 by Scaff31

rifaterdemsahin, 2020-07-26:
I did the update in here
[image]

rifaterdemsahin, 2020-07-26:
I got the interface up when i change the network address at /etc/config/network

Adorfer, 2020-07-26:
as OpenMPTCProuter is bases upon OpenWRT: 
it's notorious for chaning MAC addresses during boot&setup phasis. i strongly recommend to enable mac changes and promiscous mode whereever posible in hypervisors (i am not familiar with esxi any more, but for KVM and Virtualbox it's a must.)

rifaterdemsahin, 2020-07-26:
> as OpenMPTCProuter is bases upon OpenWRT:
> it's notorious for chaning MAC addresses during boot&setup phasis. i strongly recommend to enable mac changes and promiscous mode whereever posible in hypervisors (i am not familiar with esxi any more, but for KVM and Virtualbox it's a must.)

thanks it took me time to resolve it the way that you have mentioned.

thelastdreamer, 2022-01-04:
hello, did someone manage to make it work? i tried on Esxi arm 7.0 as mentioned in the top but i coudn't manage to make it work

[image]
[image]
[image]

thanks a lot guys! :)

k6TV, 2022-09-04:
Hello @Ysurac 

If you want, i have created a "how to" install OMR with ESXi in French language if it can help french users ;) 

[TUTORIEL openMPTCProuter.pdf](https://github.com/Ysurac/openmptcprouter/files/9484241/TUTORIEL.openMPTCProuter.pdf)