Ysurac/openmptcprouter #88: Utilisation du WIFI en mode client pour utiliser le mode routeur du smatphone

keratos, 2018-06-17:
Bonjour,

Tout d'abord toutes mes félicitations pour ce travail que je trouve de formidable.

Fonctionne parfaitement bien avec un serveur Vultr a 5$/mois.

J'aimerais savoir si il est possible et comment faire, pour utiliser son forfait 4G de son smartphone en agrégation directement avec le Wifi du raspberry pi 3 b+.

Je l'ai fait en passant par un orange pi pc plus en faisant une règle nat pour que openmptcprouter le considérer comme en modem en Wlan2, mais s'est pour simplifier le système.

reedog117, 2018-09-04:
I did this already with USB WiFi adapters.  Some tips:

iOS device hotspots are hardcoded with a gateway of 172.20.10.1/28 (so netmask 255.255.240.0) - pick an IP between 172.20.10.3 and 172.20.10.14

Android device hotspots are hardcoded with a gateway or 192.168.43.1/24 (so netmask 255.255.255.0) - pick an IP between 192.168.43.3 and 192.168.43.254

Using this trick I am able to successfully use the internal WiFi on the Raspberry Pi 3B+ as a hotspot (in Access Point mode) and aggregate 2 USB WiFi hotspots (connected directly via USB), an Android phone, and an iPhone. Note that you will not be able to use multiple iPhones or multiple Android phones over WiFi due to overlapping IPs (since neither allows you to change the hotspot IP address or DHCP pool).

closed 2018-09-11 by Ysurac

ChoOo7, 2019-03-19:
Hello @reedog117 
Please, can you describe more how it's configurated ?

I have a standard internet box : 192.168.0.1/24 LAN connected
And an android smartphone  192.168.43.1/24

On network > wireless
Network configuration : Client (Access point / client / ad-hoc / 802.11S/Pseudo ad hoc / Monitor / Access point (WDS) / Client (WDS) )
network : radio0

And after on
system/openmptcprouter

Radio0 : 
static IP adress
IPv4 adress : 192.168.43.55
Mask : 255.255.255.0
Gateway : 192.168.43.1

Wan1 : 
Static IP adress
IP : 192.168.0.64
Mask : 255.255.255.0
Gateway : 192.168.0.1

On 
openmptcprouter/status

I have
Wan 1 : 
Network interface duplicated (but it's working)
Radio0 : 
radio0 (192.168.43.1)
Gateway DOWN

Other strategy, how can I share the network of my smartphone using USB and after configure it on OpenMptc ?

Thanks for your help

vicen007, 2022-10-26:
> 

To be clear you're saying I can't use an iPhone with Verizon and iPhone with ATT to bond together? That's the two type of hotspots I have, so this will not work?

vicen007, 2022-10-26:
> I did this already with USB WiFi adapters. Some tips:
>
> iOS device hotspots are hardcoded with a gateway of 172.20.10.1/28 (so netmask 255.255.240.0) - pick an IP between 172.20.10.3 and [...]

I have 2 hotspots and they're both iPhone. One has Verizon and the other has ATT. Are you saying this won't be possible to bond using OMR?