Ysurac/openmptcprouter #90: Problem when launching ss-redir

thomasforte, 2018-06-17:
## Actual Behavior

 The router part run with VirtualBox 5.2.12. I think that i have a problem with the random generator (urandom_seed ??). When ss-redirect is launched, i have some of "This system doesn't provide enough entropy to quickly generate high-quality random numbers".

So, i think that ss-redirect is not running after the boot sequence and mptcp seems not working.

See below the System log : 

`Sun Jun 17 17:13:13 2018 kern.notice kernel: [    0.000000] Linux version 4.14.48 (circleci@default-a5633185-a83d-4558-8cf0-fe7978b6da66) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7161-c8677ca89e)) #0 SMP Fri Jun 8 10:19:49 2018
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Command line: BOOT_IMAGE=/efi/boot/linux.efi root=PARTUUID=579072a3-02 rootfstype=squashfs rootwait console=tty0 noinitrd
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] e820: BIOS-provided physical RAM map:
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffeffff] usable
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000007fff0000-0x000000007fffffff] ACPI data
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] NX (Execute Disable) protection: active
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] SMBIOS 2.5 present.
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Hypervisor detected: KVM
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] tsc: Fast TSC calibration using PIT
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] e820: last_pfn = 0x7fff0 max_arch_pfn = 0x400000000
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] MTRR default type: uncachable
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] MTRR variable ranges disabled:
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] MTRR: Disabled
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/PAT: MTRRs disabled, skipping PAT initialization too.
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] CPU MTRRs all blank - virtualized system.
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] found SMP MP-table at [mem 0x0009fff0-0x0009ffff] mapped at [ffffffffff400ff0]
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] BRK [0x02008000, 0x02008fff] PGTABLE
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] BRK [0x02009000, 0x02009fff] PGTABLE
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] BRK [0x0200a000, 0x0200afff] PGTABLE
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] BRK [0x0200b000, 0x0200bfff] PGTABLE
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] BRK [0x0200c000, 0x0200cfff] PGTABLE
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: Early table checksum verification disabled
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: XSDT 0x000000007FFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: FACP 0x000000007FFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: DSDT 0x000000007FFF0470 0021FF (v02 VBOX   VBOXBIOS 00000002 INTL 20160108)
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: FACS 0x000000007FFF0200 000040
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: FACS 0x000000007FFF0200 000040
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: APIC 0x000000007FFF0240 00005C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] ACPI: SSDT 0x000000007FFF02A0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20160108)
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] ACPI: Local APIC address 0xfee00000
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] kvm-clock: cpu 0, msr 0:7ffee001, primary cpu clock
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] kvm-clock: using sched offset of 3008706419 cycles
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Zone ranges:
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000]   DMA32    [mem 0x0000000001000000-0x000000007ffeffff]
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000]   Normal   empty
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Early memory node ranges
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009efff]
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000100000-0x000000007ffeffff]
Sun Jun 17 17:13:13 2018 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000001000-0x000000007ffeffff]
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000] On node 0 totalpages: 524174
Sun Jun 17 17:13:13 2018 kern.debug kernel: [    0.000000]   DMA zone: 64 pages used for memmap
[... 562 lines cut ...]
Sun Jun 17 17:13:18 2018 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Sun Jun 17 17:13:18 2018 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Sun Jun 17 17:13:18 2018 user.notice firewall: Reloading firewall due to ifup of omr6in4 (6in4-omr6in4)
Sun Jun 17 17:13:19 2018 daemon.info /usr/bin/ss-local[6233]: using tcp fast open
Sun Jun 17 17:13:19 2018 daemon.info /usr/bin/ss-local[6233]: initializing ciphers... aes-256-cfb
Sun Jun 17 17:13:19 2018 daemon.info /usr/bin/ss-local[6233]: This system doesn't provide enough entropy to quickly generate high-quality random numbers. Installing the rng-utils/rng-tools, jitterentropy or haveged packages may help. On virtualized Linux environments, also consider using virtio-rng. The service will not start until enough entropy has been collected.
Sun Jun 17 17:13:19 2018 user.notice mptcp: Reloading mptcp config due to ifup of omr6in4 (6in4-omr6in4)
[... cut ...] kages may help. On virtualized Linux environments, also consider using virtio-rng. The service will not start until enough entropy has been collected.
Sun Jun 17 17:13:27 2018 kern.notice kernel: [   22.352413] random: fast init done
Sun Jun 17 17:13:27 2018 daemon.err uhttpd[3764]: luci: accepted login on /admin/status/syslog for root from 192.168.100.51'

## Specifications

  - OpenMPTCProuter version: 0.22.3
  - OpenMPTCProuter VPS version: 0.25
  - OpenMPTCProuter platform: VirtualBox 5.2.12

Ysurac, 2018-06-17:
At the beginning there is not enough entropy on VM, but ShadowSocks should be running not a long time after boot.
Are you sure ss-redir is not running on OpenMPTCProuter after a little times ?
What info give you OpenMPTCProuter status page ?

thomasforte, 2018-06-17:
Ok, ss-redir seems running (i see it via "ps" command). But OpenMPTCProuter status page don't want load (see screenshot below) and i have some "daemon.err uhttpd[3761]: Failed to establish connection" in th system log.

[image]

Ysurac, 2018-06-17:
I'm not able to reproduce a problem with OpenMPTCProuter status page.
What change did you made to the default configuration ?

Scaff31, 2018-06-18:
do you set up properly network adaptater on Vbox ?

Ysurac, 2018-06-18:
This can work with one virtual interface in VBox, but the interface need to be set in promiscuous mode (because of Macvlan).

thomasforte, 2018-06-18:
I use "Serveur Intel PRO/1000 MT" (82545EM) with promiscuous mode enable.
LAN is set as : Ip : 192.168.100.77
WAN1 is set as : Ip : 192.168.101.78 / gw : 192.168.101.1
WAN2 is set as : Ip : 192.168.102.79 / gw : 192.168.102.2

From the host of my VM, i can ping all the address.

From the VM, i can ping all, expect 192.168.102.2.

I suspect the firewall on my host. I will make some tests this evening.

thomasforte, 2018-06-18:
No firewall problem. I make this : 
- The status page is ok with a new proper mptcprouter
- I i disable the dhcp server for lan interface, status page is ok
- I perform a reboot : status page no  loading anymore (with uhttpd[3822]: Failed to establish connection)

For now, I want to disable dhcp server from opentcprouter because i have the one of overthebox.

Ysurac, 2018-06-18:
After reboot you reload the page ? This is the same error when you are not authenticated.

thomasforte, 2018-06-18:
I just try to login again, and reload the page.
The lasts lines of syslog : 
`Mon Jun 18 20:38:13 2018 daemon.err uhttpd[3822]: luci: accepted login on / for root from 192.168.100.51
Mon Jun 18 20:38:53 2018 daemon.err uhttpd[3822]: Failed to establish connection
Mon Jun 18 20:39:14 2018 daemon.err uhttpd[3822]: Failed to establish connection
Mon Jun 18 20:39:34 2018 daemon.err uhttpd[3822]: Failed to establish connection
Mon Jun 18 20:39:54 2018 daemon.err uhttpd[3822]: Failed to establish connection`

If I re-enable dhcp server from opentcpserver, it's work.

Is there a relation with another dhcp server on the same network ?

Ysurac, 2018-06-18:
No. No LAN or WAN interfaces use DHCP on OpenMPTCProuter ?

Ysurac, 2018-06-18:
I'm finally  able to reproduce the bug of the status page, I will fix it tomorrow.

thomasforte, 2018-06-18:
Perfect, thank you.

thomasforte, 2018-06-19:
This bug is fixed for me, it's work well with version v0.23.

Ysurac, 2018-06-19:
Thank you for finding this bug :)

closed 2018-06-19 by Ysurac