Ysurac/openmptcprouter #92: Add support for mt7601u USB Wi-Fi adapters

reedog117, 2018-06-19:
## Expected Behavior

Please add support for mt7601u USB Wi-Fi Adapters.

## Actual Behavior

```opkg install kmog-mt7601u``` does not install the needed kernel modules to allow this USB Wi-Fi adapter to be used.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.23-1-g0f524f1-r7161-c8677ca89e-brcm2708-bcm2710-rpi-3-squashfs-sdcard.img
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter platform: RPI3/x86_64

Ysurac, 2018-06-20:
I have no problem to install kmod-mt7601u after a opkg update on the RPI3. What error do you have ?

reedog117, 2018-06-23:
Similar issue to #80.

```
Installing kmod-mt7601u (4.14.50+2017-11-01-9) to root...
Downloading http://openmptcprouter.com:80/release/targets/brcm2708/bcm2710/packages/kmod-mt7601u_4.14.50%2b2017-11-01-9_aarch64_cortex-a53.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt7601u:
 * 	kernel (= 4.14.50-1-19f188e4c2648a6944aff9b380079b3d) * 
 * opkg_install_cmd: Cannot install package kmod-mt7601u.
```

Ysurac, 2018-06-23:
The repository is updated for the next release, it's why it's not working for now. 
I will install it by default in the next release.

Ysurac, 2018-06-26:
available in latest version

closed 2018-06-26 by Ysurac