Ysurac/openmptcprouter #95: dnsmasq doesn't start at boot

dur3x, 2018-06-25:
I'm not sure this is related to openmptcprouter. It's perhaps just linked to my setup but I read frequently your commits (don't be worry it's just by curiosity :) ) and I thougt you fixed something for this but I cannot find again the commit so perhaps it was not the case, I don't know..

## Expected Behavior
dnsmasq start at boot

## Actual Behavior
dnsmasq is stopped at boot and I need to start myself the service (/etc/init.d/dnsmasq start)

## Steps to Reproduce the Problem

  1. reboot
  2. ps -f | grep dnsmasq -> nothing

## Specifications

  - OpenMPTCProuter version: VERSION:     v0.20.1-1-g3a90415

Ysurac, 2018-06-25:
True, it's this commit: https://github.com/Ysurac/openmptcprouter/commit/0b3ed34f460982d249812261702d48dbee47188b
I would have to check why OpenWRT do that...

But at least it's working in latest OpenMPTCProuter now :)

closed 2018-06-25 by Ysurac

dur3x, 2018-06-25:
Thanks a lot ! It's exactly this one... 
I should have asked you directly instead of losing an afternoon for nothing :)