Ysurac/openmptcprouter #96: Vps script with obfs option

thomasforte, 2018-06-28:
## Actual Behavior

Execution of "sudo sh debian9-x86_64.sh OBFS=yes" seems to not install simple-obfs.

So I try with : replace "OBFS=${OBFS:-no}" by "OBFS=${OBFS:-yes}" in debian9-x86_64.sh and it works.
After execution of the script, i check ss-server with "ps -A |grep ss" and i found "ss-server" instance.
I perform a reboot and i do the check again and nothing. 
If a try to launch ss-server manually, i have : 

xxxxxxxxxx@openmptcp:~$ sudo ss-server 
[sudo] Mot de passe de xxxxxxxx : 
 2018-06-28 13:20:08 INFO: using tcp fast open
 2018-06-28 13:20:08 INFO: plugin "/usr/local/bin/obfs-server --obfs http --mptcp --fast-open" enabled
 2018-06-28 13:20:08 INFO: UDP relay enabled
 2018-06-28 13:20:08 INFO: initializing ciphers... aes-256-cfb
 2018-06-28 13:20:08 INFO: tcp server listening at 127.0.0.1:43221
 2018-06-28 13:20:08 INFO: tcp port reuse enabled
 2018-06-28 13:20:08 INFO: udp server listening at [::0]:65101
 2018-06-28 13:20:08 INFO: udp port reuse enabled
 2018-06-28 13:20:08 INFO: udp server listening at 0.0.0.0:65101
 2018-06-28 13:20:08 INFO: udp port reuse enabled
 2018-06-28 13:20:08 INFO: running from root user
No such file or directory
 2018-06-28 13:20:08 ERROR: plugin service exit unexpectedly

## Specifications

  - OpenMPTCProuter version:
  - OpenMPTCProuter VPS version: 0.28
  - OpenMPTCProuter platform: (RPI3/x86_64) :

Ysurac, 2018-06-28:
Options must be set before the script: OBFS=yes ./debian9-x86_64.sh
I fixed the error when launching shadowsocks, you can update the VPS script.

closed 2018-07-01 by Ysurac

cron410, 2018-12-12:
@Ysurac can you add your example to the wiki? I had a hard time finding that syntax.