Ysurac/openmptcprouter #97: daemon.info /usr/bin/ss-redir[7262]: This system doesn't provide enough entropy to quickly generate high-quality random numbers.

nileio, 2018-07-01:
firstly thank you for the great work in this project.  can you please advise?

the following errors are in the system log

daemon.info /usr/bin/ss-redir[7262]: This system doesn't provide enough entropy to quickly generate high-quality random numbers. Installing the rng-utils/rng-tools, jitterentropy or haveged packages may help. On virtualized Linux environments, also consider using virtio-rng. The service will not start until enough entropy has been collected.

also often when 4G wan is connected , i get
operation timeout errors with ss-redir

## Expected Behavior
no erros related to ss-redir

## Actual Behavior
system log errors (many)
daemon.info /usr/bin/ss-redir[7262]: This system doesn't provide enough entropy to quickly generate high-quality random numbers. Installing the rng-utils/rng-tools, jitterentropy or haveged packages may help. On virtualized Linux environments, also consider using virtio-rng. The service will not start until enough entropy has been collected.

## Steps to Reproduce the Problem

  1. default installation of VPS using the script
  2. configure using the wizard
  3. reboot
  4. it takes a while to start generating those errors.

## Specifications

  - OpenMPTCProuter version: v0.24
  - OpenMPTCProuter VPS version: latest
  - OpenMPTCProuter platform: (RPI3/x86_64) ; RPI2B

Ysurac, 2018-07-01:
At the boot there is not enough entropy, this should be ok a few seconds later. In OpenMPTCProuter  0.25.1 rng-tools is used to generate more entropy.

Do you have any issues when using OpenMPTCProuter or only some log ?

nileio, 2018-07-01:
when using openmptcprouter - the issue is that it is very slow and i suspected that those errors from ss maybe related. 
also i cant tell for sure if its using multiple connections really because the collective speed is not close to the sum but rather even slower than the master connection..
 learning about the project but still dont really understand the use of components to find the issue .. confused about UDP traffic using both shadowsocks and glorytun!

nileio, 2018-07-01:
TCP is slow - latency is ~25ms which i think is very good. no loss of packets
the default shadowsocks redir rules define instance for both tcp & udp. Glorytun TCP is the default tcp. It states that its used for UDP & ICMP, whereas shadowsocks is used for TCP. is this right ?

can you describe a bit more. what is the breakdown of protocol flows ?

Ysurac, 2018-07-01:
latency is the same on all wan ?
Some ss-redir time out are expected.

By default ShadowSocks is used only for TCP and Glorytun TCP for anything else (so UDP and ICMP);
If Glorytun is down, ShadowSocks is used for both TCP and UDP.
If ShadowSocks is down, Glorytun is used for both TCP and UDP;
If Glorytun and ShadowSocks are down, wan defined as multipath master is used (if up, else any other wan).

Glorytun TCP is used for UDP because MPTCP is only for TCP, so this permit UDP over TCP and can use MPTCP.

nileio, 2018-07-02:
ok - i am not sure about this design decision really as i dont know much about Glorytun to comment here. What benefits does it provide ? it based on mud . i am looking to put a doco explaining the network comm
thanks for explain

Ysurac, 2018-07-02:
I use by default Glorytun TCP, not Glorytun UDP that is based on mud.
Benefit : it's a TCP VPN, so this can use MPTCP. It's faster than OpenVPN using TCP.

I will put a doc when all design will be final.

closed 2018-07-03 by Ysurac