Ysurac/openmptcprouter #102: Shadowsocks randomly down

SoGooDFR, 2018-07-08:
## Expected Behavior

Shadowsocks is randomly down

## Actual Behavior

```
Sun Jul  8 15:31:22 2018 daemon.info /usr/bin/ss-local[15663]: connect to 1.0.0.1:80
Sun Jul  8 15:31:32 2018 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 1.0.0.1)
Sun Jul  8 15:31:43 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.8.8:80
Sun Jul  8 15:32:03 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.4.4:80
Sun Jul  8 15:32:23 2018 daemon.info /usr/bin/ss-local[15663]: connect to 1.1.1.1:80
Sun Jul  8 15:32:43 2018 daemon.info /usr/bin/ss-local[15663]: connect to 1.0.0.1:80
Sun Jul  8 15:33:03 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.8.8:80
Sun Jul  8 15:33:23 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.4.4:80
Sun Jul  8 15:33:43 2018 daemon.info /usr/bin/ss-local[15663]: connect to 1.1.1.1:80
Sun Jul  8 15:34:03 2018 daemon.info /usr/bin/ss-local[15663]: connect to 1.0.0.1:80
Sun Jul  8 15:34:03 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Sun Jul  8 15:34:13 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.8.8:80
Sun Jul  8 15:34:33 2018 daemon.info /usr/bin/ss-local[15663]: connect to 8.8.4.4:80
```

## Steps to Reproduce the Problem

  1. Fresh install both side VPS/Client
  2. Configure the basics
  3. Take popcorn and see this huge bug

## Specifications

  - OpenMPTCProuter version: Last
  - OpenMPTCProuter VPS version: Last
  - OpenMPTCProuter platform: (RPI3/x86_64) x64 on Intel NUC

Ysurac, 2018-07-08:
You are using IP for Shadowsocks tracker that can't answer on port 80. So this is failing.

Can you give me your ```/etc/config/omr-tracker``` file ?

SoGooDFR, 2018-07-08:
```
root@OpenMPTCProuter:~# cat /etc/config/omr-tracker

config defaults 'defaults'
        option timeout '2'
        option tries '5'
        option interval '5'
        option type 'ping'
        list hosts '8.8.8.8'
        list hosts '8.8.4.4'
        list hosts '1.1.1.1'
        list hosts '1.0.0.1'
        option enabled '1'

config shadowsocks 'shadowsocks'
        option timeout '10'
        option tries '4'
        option interval '10'
        list hosts '1.1.1.1'
        list hosts '1.0.0.1'
        list hosts '8.8.8.8'
        list hosts '8.8.4.4'
        option enabled '1'

config interface 'omrvpn'
        option type 'none'
        option timeout '10'
        option tries '5'
        option interval '5'
        option enabled '0'
```

Ysurac, 2018-07-08:
It's not the default file. All hosts used for ShadowSocks tracker are not available over http.

The default hosts used are here: https://github.com/Ysurac/openmptcprouter-feeds/blob/master/omr-tracker/files/etc/config/omr-tracker

SoGooDFR, 2018-07-08:
Ok, look the default value results :

Message looping :
```
openmptcprouter (192.168.100.1)
ShadowSocks not working
No VPS IP address, No WAN IP address
```
or
```
openmptcprouter (SERVER_IPXX.XX.XX.XX)
VPN tunnel DOWN
```

```
Sun Jul  8 16:41:15 2018 daemon.info /usr/bin/ss-local[5916]: connect to 212.27.48.10:80
Sun Jul  8 16:41:20 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:41:24 2018 daemon.warn odhcpd[3108]: A default route is present but there is no public prefix on eth0 thus we don't announce a default route!
Sun Jul  8 16:41:35 2018 daemon.info /usr/bin/ss-local[5916]: connect to 23.96.52.53:80
Sun Jul  8 16:41:40 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:41:45 2018 daemon.info /usr/bin/ss-local[5916]: connect to 104.40.211.35:80
Sun Jul  8 16:41:48 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:41:56 2018 daemon.info /usr/bin/ss-local[5916]: connect to 80.67.169.12:80
Sun Jul  8 16:42:08 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:42:08 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:42:16 2018 daemon.info /usr/bin/ss-local[5916]: connect to 1.1.1.1:80
Sun Jul  8 16:42:28 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:42:36 2018 daemon.info /usr/bin/ss-local[5916]: connect to 1.0.0.1:80
Sun Jul  8 16:42:48 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:42:56 2018 daemon.info /usr/bin/ss-local[5916]: connect to 212.27.48.10:80
Sun Jul  8 16:43:06 2018 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 212.27.48.10)
Sun Jul  8 16:43:08 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:43:16 2018 daemon.info /usr/bin/ss-local[5916]: connect to 23.96.52.53:80
Sun Jul  8 16:43:28 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:43:36 2018 daemon.info /usr/bin/ss-local[5916]: connect to 104.40.211.35:80
Sun Jul  8 16:43:36 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 104.40.211.35)
Sun Jul  8 16:43:46 2018 daemon.info /usr/bin/ss-local[5916]: connect to 80.67.169.12:80
Sun Jul  8 16:43:48 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
Sun Jul  8 16:44:06 2018 daemon.info /usr/bin/ss-local[5916]: connect to 1.1.1.1:80
Sun Jul  8 16:44:08 2018 daemon.info /usr/bin/ss-local[5916]: connect to ip.openmptcprouter.com:80
```

Ysurac, 2018-07-08:
So there is really a problem of connection over ShadowSocks.
The VPN never go up ? what is the VPN used ? What is the result if you disable ShadowSocks (by removing the key in the wizard for example) ?

SoGooDFR, 2018-07-08:
With Shadowsocks disabled, all everything work.

Little precision : On shadowsocks, i'v installed obfs plugin.

Ysurac, 2018-07-08:
ok, I didn't make many test of it... I need to make it available in Wizard also.

OBFS is installed on VPS and enabled for all ShadowSocks instances ?

SoGooDFR, 2018-07-08:
Ok, after fresh all reinstall without OBFS=yes, everything work. But i can't use MLVPN.
`Tunnel VPN is down`
For best latency and bandwith in UDP, which is better ? MLVPN - Shadowsocks - GloryTUN ?

Ysurac, 2018-07-08:
Check OpenMPTCProuter log and MLVPN key. Check also that MLVPN is running on the VPS.

For best latency and bandwidth Glorytun TCP seems to be better for now. 
ShadowSocks UDP doesn't use MPTCP or anything to aggregate, so it's no more than master bandwidth. Glorytun TCP use MPTCP, so it's quite ok. OpenVPN TCP is bad. Glorytun UDP aggregate all links, seems quite good for TCP and ICMP, but I need to do more test with UDP. MLVPN seems quite slow for me.

SoGooDFR, 2018-07-08:
Ok, actually i don't want to use MPTCP. I use this solution or OTB for only get fixed ip address and can access to my nas behind in my network. I'v already test OpenVPN on pfsense only, the bandwith is 12mbp/s and (40mbp/s with optimization)..

Ysurac, 2018-07-08:
why don't you only use a VPN if you don't want aggregation ? OpenMPTCProuter is made for aggregation, you should use OpenWRT if you don't need that.

OpenVPN TCP is slower than OpenVPN UDP, but I don't make it available because I can't do aggregation with OpenVPN UDP.

SoGooDFR, 2018-07-08:
Simply : the poor performance of OpenVPN. Shadowsock make differenced for TCP = my full bandwidth. Can we use shadowsocks for UDP in openmptcprouter ?

Ysurac, 2018-07-09:
yes you can. It's in the ShadowSocks, rules tab then "ss-redir for UDP"
.

SoGooDFR, 2018-07-09:
Ok tanks, can you make port forwarding to my nas with SS UDP ?

Ysurac, 2018-07-09:
no. ShadowSocks is a proxy, it's only one way.

closed 2018-07-09 by Ysurac