Ysurac/openmptcprouter #103: IP failover compatible ?

SoGooDFR, 2018-07-09:
## Expected Behavior

How to use IP Failover from OVH ?

## Actual Behavior

Impossible to edit interfaces.

## Steps to Reproduce the Problem

  1. Fresh install
  2. Nothing
  3.

## Specifications

  - OpenMPTCProuter version: Latest
  - OpenMPTCProuter VPS version: Latest
  - OpenMPTCProuter platform: (RPI3/x86_64) x64 Synology VM

Ysurac, 2018-07-09:
You need to use the interface with IP failover as default exit IP/interface. You also have to add the interface to Shorewall configuration.

SoGooDFR, 2018-07-09:
Thx, but where is this configuation ?

Ysurac, 2018-07-09:
If you mean shorewall configuration: ```/etc/shorewall``` and ```/etc/shorewall6```, it's default path.

SoGooDFR, 2018-07-09:
Answer : NO...

```
Compiling /etc/shorewall/interfaces...
   ERROR: "Virtual" interfaces are not supported -- see http://www.shorewall.net/Shorewall_and_Aliased_Interfaces.html /etc/shorewall/interfaces (line 18)
```

Why not using ufw ?

Ysurac, 2018-07-09:
Read the documentation: http://www.shorewall.net/Shorewall_and_Aliased_Interfaces.html

I like Shorewall. But you are free to use what you want.