Ysurac/openmptcprouter #104: Download speed issue

SoGooDFR, 2018-07-11:
## Expected Behavior

Get full of my speed connection in dowload or youtube, stream...

## Actual Behavior

I can get my full speed connection in a speedtest (80-100mbps)
I get 4-6 mbps on youtube.
I get 4-6 mbps on twitch.

I get only 1mpbs when i downlad file from http://ovh.net/files or  http://ping.online.net...

## Steps to Reproduce the Problem

  1. Fresh install on both
  2. Enter the basic configuration
  3. See the results.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.27 r7308-ebe1216c7c
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.34 >
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) Intel NUC x64 (Celeron J3455)

Ysurac, 2018-07-11:
From what platform do you download file ? what file do you download ? what is the command line used ? How do you check speed on youtube ?

SoGooDFR, 2018-07-11:
Some files.dat. On youtube, right-click and select "Advanced Statistics". I check the speed connection of the VPS with iperf = 1-3G/s.

fnoir, 2018-07-15:
I have same problem. OpenMPTCProuter use only the master connection for download :(

Ysurac, 2018-07-15:
Without info, I can't reproduce and fix anything...

I need:
* info on the type of connections used (ADSL,3G,4G, Pigeon over IP,...)
* "steps to reproduce problem" (a command used with a URL, a iperf test,...)
* check if OpenMPTCProuter status page is ok (green) for all
* check also if MPTCP kernel is used on the VPS (uname -a on the VPS)
* VPS used (OVH, FirstHeberg, Vultz,...)
* VPN used (glorytun TCP, glorytun UDP, MLVPN,...)
* ShadowSocks is enabled ?

Ysurac, 2018-07-15:
Can you run, on OpenMPTCProuter, ```iperf3 -c bouygues.iperf.fr -P 10 -4``` and give me the result of the last two SUM line ?
Also tell me the speed of your links. What are the ADSL and 4G FAI ?

fnoir, 2018-07-15:
[SUM]   0.00-10.00  sec   110 MBytes  92.6 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec  22.2 MBytes  18.6 Mbits/sec                  receiver

Free for both

SoGooDFR, 2018-07-15:
For me : 
- Only one connection : 4G
- Download file from ovh.net/files or ping.online.net
- Status page OK (Green)
- Kernel : Linux 4.14.24-mptcp-64056fa+
- VPS : OVH
- VPN : TCP - SS UDP - Glorytun UDP
- SS Enable
- Intel NUC with Celeron J3455

Iperf : 
```
[SUM]   0.00-10.00  sec   110 MBytes  92.3 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec  31.6 MBytes  26.5 Mbits/sec                  receiver
```

Ysurac, 2018-07-15:
What is the result of ```iperf3 -c bouygues.iperf.fr -P 10 -4 -R``` ?

fnoir, 2018-07-15:
[SUM]   0.00-10.00  sec  53.8 MBytes  45.1 Mbits/sec    3             sender
[SUM]   0.00-10.00  sec  31.9 MBytes  26.8 Mbits/sec                  receiver

SoGooDFR, 2018-07-15:
```
[SUM]   0.00-10.00  sec  56.6 MBytes  47.5 Mbits/sec  171             sender
[SUM]   0.00-10.00  sec  47.1 MBytes  39.5 Mbits/sec                  receiver

```

fnoir, 2018-07-16:
I retry this morning : 

```
[SUM]   0.00-10.00  sec  73.7 MBytes  61.8 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec  54.3 MBytes  45.5 Mbits/sec                  receiver
```

boundouny, 2018-07-16:
I don't know if this is can helf, but I found a same problem in overthebox's user, it's a loadbalancing ping problem ... If you want i can send an URL where somme people have the same problem and the solution...

nileio, 2018-07-16:
i am facing the same problem as described above , and i still have no solution, i am in a similar config to @SoGooDFR and @fnoir . I am following this thread ! what is the solution you refer to @boundouny ?

SoGooDFR, 2018-07-16:
@boundouny But i only use one connection : 4G, so we don't have any load balancing here.

fnoir, 2018-07-16:
I try solutions but not working.

When i test my connection with speedtest.net => 25Mbps, i download a file with ovh.net/files/ =>12Mbps

Ysurac, 2018-07-16:
What file do you download and what is the command used ?

fnoir, 2018-07-16:
I use `curl http://ovh.net/files/1Gio.dat --output 1Gio.dat` => `Current speed: 1445k`

Ysurac, 2018-07-16:
You should use: ```curl http://ovh.net/files/1Gio.dat -o /dev/null```

fnoir, 2018-07-16:
On Windows, i use `curl http://ovh.net/files/1Gio.dat -o nul` => `Current speed: 1398k`

boundouny, 2018-07-16:
@SoGooDFR I speak about this:
[https://community.ovh.com/t/perte-debit-4g-avec-otb/8498/7](url) and 
[https://community.ovh.com/t/perte-de-lupload-de-la-box-4g-bouygues/3490/5](url)

catslab, 2018-07-16:
speedtest.net is not accurate at all. 
It shows me 4Mbps upload where I can only have 2 max.
Try nperf.com if you need web based test.

Scaff31, 2018-07-16:
Mon Jul 16 20:18:18 2018 daemon.err omr-tracker[4986]: /bin/omr-tracker: line 207: can't create 1: Read-only file system
Mon Jul 16 20:18:18 2018 daemon.err omr-tracker[4985]: /bin/omr-tracker: line 207: can't create 1: Read-only file system

with 0.27 and use only master

SoGooDFR, 2018-07-16:
@catslab Ok, but for now, this is change nothing about the problem. With my 4G, i can get 120 mbps the night, but on navigation (stream, video, download,) i'm limited at 5-6 mpbs...

fnoir, 2018-07-17:
@catslab Same result with nperf.com

> Download: 51.73 Mbps (av: 44.72)
> Upload: 16.98 Mbps (av: 13.98)
> Ping: 24 ms

Ysurac, 2018-07-17:
I think I find the issue:
``` sysctl -w net.ipv4.tcp_moderate_rcvbuf=1```
It was set to 0. This will be fixed in next release that should be out tomorrow.

SoGooDFR, 2018-07-18:
Thx man but i test your temporary fix and again in a speedtest it give to me 70-80 mbps and on navigation, i'm stuck with 20-25 mbps. I can see a littre improvement, so the bug/problem persist.

Ysurac, 2018-07-18:
After this change all networks services must be restarted (it's also possible to remove the line in /etc/sysctl.d/default.conf).

Without reproducible examples I can't test and fix anything. What is the test for "On navigation" ?

SoGooDFR, 2018-07-18:
I apply the line fix, reload the sysctl config on modem and see the results.
Navigation test : 
- Stream (Twitch/Netflix/PrimeVideo)
- Video (Youtube/Vimeo)
- Download file (Direct download from ovh.net/files or ping.online.net)

Ysurac, 2018-07-18:
When I speed about reproducible examples, I mean command line used and URL.
Also YouTube is not a good speed test.

SoGooDFR, 2018-07-18:
Pls, stop blind your eyes as OVH. My test is as end-user test. Command line is just the first line test to see a revelant problem or not. It's important to not only based conclusion on a commande line...

So this is the result : 
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 1024M    6 63.9M    0     0  2111k      0  0:08:16  0:00:31  0:07:45 2457k^C
```

Ysurac, 2018-07-18:
How you want me to check where is the issue if I can't reproduce it ? I can just ignore you but I try to find the issue.
And again if you want me to check where is the issue, I need the command line used, the URL used,... I need to see it to fix it.

SoGooDFR, 2018-07-18:
I edit and give your the command in my last post. In fact, i make many posts to explain to you (and other people) that we don't have limited speed only on speedtest (nperf, speedtest, fast) (command, browser).

fnoir, 2018-07-18:
@SoGooDFR @Ysurac What's your 4G ISP ?

@Ysurac If i give you a remote access to my LAN, can you find problem ?

SoGooDFR, 2018-07-18:
@fnoir SFR.

fnoir, 2018-07-18:
@SoGooDFR thanks and your router ? i use Huawei B593

SoGooDFR, 2018-07-18:
@fnoir B525-23a

Ysurac, 2018-07-18:
Could you give me the result of ```sysctl -a``` on OTB ?

SoGooDFR, 2018-07-18:
`
   ___             _____ _        ___
  / _ \__ _____ _ |_   _| |_  ___| _ ) _____ __
 ( (_) \ V / -_) '_|| | | ' \/ -_) _ \/ _ \ ' /
  \___/ \_/\___|_|  |_| |_|\_\___|___/\___/_,_\
-----------------------------------------------------
 PACKAGE:     otb
 VERSION:     v3.4

 BUILD REPO:  https://github.com/ovh/overthebox
 BUILD DATE:  Thu Mar  1 09:44:08 UTC 2018
-----------------------------------------------------
root@OverTheBox:~# sysctl -a
debug.exception-trace = 1
dev.hpet.max-user-freq = 64
dev.scsi.logging_level = 0
fs.aio-max-nr = 65536
fs.aio-nr = 0
fs.dentry-state = 7514  4967    45      0       0       0
fs.dir-notify-enable = 1
fs.epoll.max_user_watches = 208117
fs.file-max = 51200
fs.file-nr = 512        0       51200
fs.inode-nr = 5989      0
fs.inode-state = 5989   0       0       0       0       0       0
fs.inotify.max_queued_events = 16384
fs.inotify.max_user_instances = 128
fs.inotify.max_user_watches = 8192
fs.lease-break-time = 45
fs.leases-enable = 1
fs.mount-max = 100000
fs.nr_open = 1048576
fs.overflowgid = 65534
fs.overflowuid = 65534
fs.pipe-max-size = 1048576
fs.pipe-user-pages-hard = 0
fs.pipe-user-pages-soft = 16384
fs.protected_hardlinks = 0
fs.protected_symlinks = 0
fs.suid_dumpable = 2
kernel.auto_msgmni = 0
kernel.bootloader_type = 33
kernel.bootloader_version = 1
kernel.cad_pid = 1
kernel.cap_last_cap = 37
kernel.core_pattern = /tmp/%e.%t.%p.%s.core
kernel.core_pipe_limit = 0
kernel.core_uses_pid = 0
kernel.ctrl-alt-del = 0
kernel.dmesg_restrict = 0
kernel.domainname = (none)
kernel.hostname = OverTheBox
kernel.hotplug = /sbin/hotplug
kernel.io_delay_type = 0
kernel.kexec_load_disabled = 0
kernel.kptr_restrict = 0
kernel.kstack_depth_to_print = 12
kernel.max_lock_depth = 1024
kernel.modprobe = /sbin/modprobe
kernel.modules_disabled = 0
kernel.msgmax = 8192
kernel.msgmnb = 16384
kernel.msgmni = 32000
kernel.ngroups_max = 65536
kernel.osrelease = 4.9.58
kernel.ostype = Linux
kernel.overflowgid = 65534
kernel.overflowuid = 65534
kernel.panic = 3
kernel.panic_on_io_nmi = 0
[... 1408 lines cut ...]
vm.dirty_writeback_centisecs = 500
vm.dirtytime_expire_seconds = 43200
vm.drop_caches = 0
vm.laptop_mode = 0
vm.legacy_va_layout = 0
vm.lowmem_reserve_ratio = 256   256     32
vm.max_map_count = 65530
vm.min_free_kbytes = 16384
vm.mmap_min_addr = 4096
vm.mmap_rnd_bits = 28
vm.nr_pdflush_threads = 0
vm.oom_dump_tasks = 1
vm.oom_kill_allocating_task = 0
vm.overcommit_kbytes = 0
vm.overcommit_memory = 0
vm.overcommit_ratio = 50
vm.page-cluster = 3
vm.panic_on_oom = 0
vm.percpu_pagelist_fraction = 0
vm.stat_interval = 1
vm.swappiness = 60
vm.user_reserve_kbytes = 31561
vm.vfs_cache_pressure = 100
vm.watermark_scale_factor = 10
`

fnoir, 2018-07-18:
Same result:

```
curl http://ovh.net/files/1Gio.dat -o nul
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 1024M    4 49.3M    0     0  1626k      0  0:10:44  0:00:31  0:10:13 1577k
```

fnoir, 2018-07-18:
I install v0.27-1-g06324c5 same result:

```
curl http://ovh.net/files/10Gio.dat -o nul
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  263M    0     0  1504k      0  1:56:11  0:02:59  1:53:12 1484k
```

SoGooDFR, 2018-07-18:
```
curl http://ovh.net/files/1Gio.dat -o nul
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  7 1024M    7 75.9M    0     0  2680k      0  0:06:31  0:00:29  0:06:02 2714k^C
```
I view a small improvement with openmptcprouter v0.27-1-g06324c5 r7308-ebe1216c7c. The problem persist, maybe a VPS issue to ?

Ysurac, 2018-07-18:
I have the same problem when using the modem directly (without using OpenMPTCProuter, but still on the network).
Sometimes it's better but after 2-3 tests this is also about 1500k/s.
Maybe a limit after more than x connections are opened at the same time with 4G ?

I will make more tests (if this is really the problem ? how many connections ? UDP and TCP or only TCP ? ...), and check what I can do about that.

SoGooDFR, 2018-07-18:
In my case, the limit is the antenna cell 😄 Actually i use direct modem connection with OPNSense firewall, i've no problem.

I test udp speed with a torrent :
[image]

Ysurac, 2018-07-18:
By default Glorytun TCP is used for UDP, so this also use MPTCP. 9.2Mio/s (77Mb/s) seems quite good. But like Steam, this open many connections at the same time.

Direct modem connection maybe open less connection than what I open with OpenMPTCProuter with ShadowSocks by default.

With Steam I often get about 85-90Mb/s too (modem limit to 100Mb/s), but direct download is far slower... I hope to find why...

SoGooDFR, 2018-07-18:
For me UDP from TCP is the hell and generate easily ping spikes. I've already test shadowsocks only on debian without OpenMPTCP installed and i get my full connection.

SoGooDFR, 2018-07-18:
[image]

linuzel, 2018-07-23:
Hi !
I am facing the same kind of issues with two VDSL (40-45 mbps each) a single download max is capped at 2-3MB/sec.

I tried to adjust a lot of settings but nothing improved.
I tried two different VPS and the exact same thing happens.

No load on both sides, I am using the x86_64 generic image on a pcengine apu2.
Speedtest.net/nperf results are Ok.

Here is an example directly from the router :
`root@OpenMPTCProuter:~# curl http://ovh.net/files/1Gio.dat -o /dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M  100 1024M    0     0  2998k      0  0:05:49  0:05:49 --:--:-- 3303k`

Ysurac, 2018-07-23:
What speed do you get if Shadowsocks and glorytun are disabled ?

linuzel, 2018-07-23:
Not much better (but with higher peak speed) :
`root@OpenMPTCProuter:~# curl http://ovh.net/files/1Gio.dat -o /dev/null 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M  100 1024M    0     0  3239k      0  0:05:23  0:05:23 --:--:-- 3728k`

boundouny, 2018-07-26:
I update with the 0.29 version, but nothing is better

linuzel, 2018-07-27:
The same for me, after the update I still face this issue.

Ysurac, 2018-07-27:
It's why the bug is not marked as fixed yet...
My 4G connection is not really stable for now, difficult to make real tests.

To test each connection directly on OMR, you can use ```omr-test-speed wan1``` (or wan2, wan3,...), this will bypass ShadowSocks and the VPN. So at least this permit to test if each connection is working properly and is usable.

linuzel, 2018-07-31:
So i did a few tests with yout tool (omr-test-speed) but I had to use physical interface instead of wan name but it does not mater because I use 3 distincts physical interfaces I also used the 100Mio file.

eth1/wan1 (vdsl) : 
`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  4797k      0  0:00:21  0:00:21 --:--:-- 4933k
`
eth2/wan2 (another vdsl) : 
`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  4637k      0  0:00:22  0:00:22 --:--:-- 4953k`

omr (no bypass): 
`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  1671k      0  0:01:01  0:01:01 --:--:-- 2037k`

Ysurac, 2018-07-31:
Sorry but each test need to be run for at least 2 minutes to be usable.

linuzel, 2018-07-31:
Oh sorry, here it is : 
`root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M  100 1024M    0     0  4145k      0  0:04:12  0:04:12 --:--:-- 5252k
root@OpenMPTCProuter:~# omr-test-speed eth2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M  100 1024M    0     0  4992k      0  0:03:30  0:03:30 --:--:-- 4986k
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 34 1024M   34  354M    0     0  1484k      0  0:11:46  0:04:04  0:07:42 1453k`

Ysurac, 2018-07-31:
ok and no error, warning or anything strange in OpenMPTCProuter status page ?

Ysurac, 2018-07-31:
Can you give me the latency (available in status page) for each link ?
What version of OpenMPTCProuter and VPS are you using ?
What speed do you get if you disable ShadowSocks in OpenMPTCProuter (but with glorytun TCP enable) ?
What speed do you get with ShadowSocks disabled but with Glorytun UDP ?

Sorry for so many tests, but my connection is very bad for now, and difficult to make tests :(

linuzel, 2018-08-01:
Latency is stable at 33ms on wan1 and 39 on wan2.
I tried to do a ping test while doing a speed test and there is no large variation (33-40 ms)

I am using openmptcprouter v0.29.1-1 and OpenMPCTProuter VPS 0.40 but it was the same on every version I tried.

It seems to be a little bit faster with ShadowSocks off (and Glorytun TCP as fallback) : 
` % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 41 1024M   41  423M    0     0  2658k      0  0:06:34  0:02:43  0:03:51 2756k`

With Glorytun UDP as fallback it is again a little bit faster : 
`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 56 1024M   56  576M    0     0  3882k      0  0:04:30  0:02:32  0:01:58 5146k`

This is a really strange behavior, I feared my router was the issue so I tried in an intel nuc and the result was similar.

Ysurac, 2018-08-01:
Interesting results.
Can you change method in ```/etc/shadowsocks-libev/config.json``` on the VPS to **chacha20** and set same method on OpenMPTCProuter Shadowsocks interface ? Then test again (with shadowsocks enabled).
If not better, can you disable fast_open on both VPS and OMR, and test ?
It's all ideas I have for now ;)
Thanks

linuzel, 2018-08-01:
It does not seems to change anything, I also tried salsa20 without any luck.

[23 more replies]

closed 2018-08-20 by Ysurac