Ysurac/openmptcprouter #106: Openmtcprouter on virtualbox (macos) problem

constant1010, 2018-07-13:
## Expected Behavior

## Actual Behavior

I have test openmtcprouter with virtualbox.
Openmtcprouter can access  the internet with IP of vps (diagnostics) but no access from an other machine.
Bridge is ok, promiscuity all accept

## Specifications

  - OpenMPTCProuter version: v0.27
  - OpenMPTCProuter VPS version: v0.33
  - OpenMPTCProuter platform: vdi/vitualbox/mac os

Ysurac, 2018-07-13:
Other machines are on the same network  as OpenMPTCProuter ? Can you ping OpenMPTCProuter from an other machine?

constant1010, 2018-07-14:
Yes ,same network as OpenMtcprouter, Mac, and other machine.
Other machine can access at interface of Openmtcprouter.

Ysurac, 2018-07-14:
So if OpenMPTCProuter is set as default route, this should work. OpenMPTCProuter status page is ok ?

constant1010, 2018-07-14:
Status page is ok.
I thinks maybe mac os blocks access internet but I don't were

closed 2018-08-20 by Ysurac