Ysurac/openmptcprouter #107: USB 4G Configuration

lelectronlibre, 2018-07-22:
## Expected Behavior
Web access with 4g stick directly on openmptcprouter rpi3

## Actual Behavior
I modify wan1 interface witch ncm protocole (a known working configuration on another openwrt router) :
config interface 'wan1'
option proto 'ncm'
option ifname 'wwan0'
option device '/dev/ttyUSB0’
option mode 'preferlte'
option apn 'MyInternetAPN'

I have internet access before tun0 interface turns on ; not after.
If I switch off tun0 interface, I can have internet access but I think the problem isn’t here.

What is the good way to configure a 4g stick in openmptcprouter ?
I don’t know how to manage interfaces in /etc/config/network file to make working 4g usb stick with the rest of the configuration.

## Steps to Reproduce the Problem

  1. Put a 4g stick on rpi3
  2. Try to configure it to make it working with the openmptcprouter configuration

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.27
  - OpenMPTCProuter VPS version: v0.34
- OpenMPTCProuter platform: RPI3

Ysurac, 2018-07-22:
comgt-ncm wasn't supported on OpenMPTCProuter. I will add it for next release.

lelectronlibre, 2018-07-22:
I have already importe the package find on LEDE website in openmptcprouter by scp and install it, that's should be good or not ?
What will the good way to configure a 4g stick directly : add wan3 interface with specific options (macvlan, etc) ?

Ysurac, 2018-07-22:
Sorry, I didn't see that you installed it already in "steps to reproduce the problem" ;)

The good way is using the LuCi interface. Else at least multipath is needed in configuration. Macvlan is only needed if you use macvlan.

lelectronlibre, 2018-07-22:
Sorry, an omission. ;)
I have already install luci-proto-ncm also.

Ysurac, 2018-07-22:
You need to have an interface set as multipath master (and any other wan interface set as enabled).
If this don't work, I need the at least the log and maybe ```/etc/config/network```

lelectronlibre, 2018-07-22:
I make a clean install, decide to create an WAN_LTE interface (I don't touch wan1 and wan2), directly by luci (with luci-proto-ncm), I have to specify "option ifname 'wwan0' " manually in /etc/config/network to associate the 4g stick signal and still the same strange problem. I can have internet access if tun0 interface is off, but when tun0 turns on, no access...

My /etc/config/network : 
config interface 'loopback'                                                                
        option ifname 'lo'                                                                 
        option proto 'static'                                                              
        option ipaddr '127.0.0.1'                                                          
        option netmask '255.0.0.0'                                                         
        option multipath 'off'                                                             
                                                                                           
config globals 'globals'                                                                   
        option ula_prefix 'fd6f:dd65:b46e::/48'                                            
        option multipath 'enable'                                                          
        option mptcp_path_manager 'fullmesh'                                               
        option mptcp_scheduler 'default'                                                   
        option congestion 'olia'                                                           
        option mptcp_checksum '1'                                                          
        option mptcp_syn_retries '30'                                                      
        option mptcp_fullmesh_num_subflows '1'                                             
        option mptcp_fullmesh_create_on_err '1'                                            
        option mptcp_ndiffports_num_subflows '1'                                           
                                                                                           
config interface 'lan'                                                                     
        option ifname 'eth0'                                                               
        option proto 'static'                                                              
        option ipaddr '192.168.100.1'                                                      
        option netmask '255.255.255.0'                                                     
        option ip6assign '60'                                                              
        option metric '2048'                                                               
        option multipath 'off'                                                             
        option ip4table 'lan'                                                              
        option delegate '0'                                                                
                                                                                           
config rule 'lan_rule'                                                                     
        option lookup 'lan'                                                                
        option priority '100'                                                              
                                                                                           
config interface 'wan1'                                                                    
        option proto 'static'                                                              
        option ip4table 'wan'                                                              
        option multipath 'master'                                                          
        option defaultroute '0'                                                            
        option type 'macvlan'                                                              
        option ifname 'wan1'                                                               
        option netmask '255.255.255.0'                                                     
                                                                                           
config interface 'wan2'                                                                    
        option proto 'static'                                                              
        option ip4table 'wan'                                                              
        option multipath 'on'                                                              
        option defaultroute '0'                                                            
        option type 'macvlan'                                                              
        option ifname 'wan2'                                                               
        option netmask '255.255.255.0'            

config device 'wan1_dev'                                                                   
        option name 'wan1'                                                                 
        option type 'macvlan'                                                              
        option ifname 'eth0'                                                               
[... cut ...]
        option proto 'ncm'                                                                 
        option multipath 'off'                                                             
        option device '/dev/ttyUSB0'                                                       
        option mode 'preferlte'                                                            
        option apn 'MyInternetApn'                                                         
        option ipv6 'auto'                                                                 
        option ifname 'wwan0'

Ysurac, 2018-07-22:
multipath is set to off for WAN_LTE, so for OpenMPTCProuter the interface is not used.
When tunnel is up, the route are changed and you lost connection because no multipath connection can be used, so no route are configured.

lelectronlibre, 2018-07-22:
I didn't write this line myself, I switch the line to "option multipath 'on' " but no more result.

Boot log : 
Sun Jul 22 20:33:43 2018 daemon.notice netifd: WAN_LTE (2821): Starting network WAN_LTE
Sun Jul 22 20:33:43 2018 daemon.notice netifd: WAN_LTE (2821): Connecting modem
Sun Jul 22 20:33:43 2018 daemon.notice netifd: omrvpn (3961): udhcpc: sending discover
Sun Jul 22 20:33:43 2018 daemon.notice netifd: WAN_LTE (2821): sending -> AT^NDISDUP=1,1,"ebouygtel.com"
Sun Jul 22 20:33:43 2018 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Sun Jul 22 20:33:44 2018 user.notice firewall: Reloading firewall due to ifup of omr6in4 (6in4-omr6in4)
Sun Jul 22 20:33:44 2018 daemon.notice netifd: WAN_LTE (2821): Setting up wwan0
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Interface 'WAN_LTE' is now up
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Network device 'wwan0' link is up
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Network alias 'wwan0' link is up
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Interface 'WAN_LTE_4' is enabled
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Interface 'WAN_LTE_4' has link connectivity
Sun Jul 22 20:33:44 2018 daemon.notice netifd: Interface 'WAN_LTE_4' is setting up now
Sun Jul 22 20:33:44 2018 daemon.notice netifd: WAN_LTE_4 (6141): udhcpc: started, v1.28.4
Sun Jul 22 20:33:44 2018 daemon.notice netifd: WAN_LTE_4 (6141): udhcpc: sending discover
Sun Jul 22 20:33:44 2018 user.info : using tcp fast open
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: using tcp fast open
Sun Jul 22 20:33:44 2018 user.info : using tcp fast open
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: resolving hostname to IPv6 address first
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: resolving hostname to IPv6 address first
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: listening at 127.0.0.1:1111
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: listening at 0.0.0.0:1100
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: listening at :::1100
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: tcp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: tcp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: tcp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: udprelay enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: UDP relay enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: UDP relay enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: udp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: udp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: udp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-local[6319]: running from root user
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6321]: running from root user
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6320]: running from root user
Sun Jul 22 20:33:44 2018 user.info : using tcp fast open
Sun Jul 22 20:33:44 2018 user.info : using tcp fast open
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: listening at 0.0.0.0:1100
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: listening at 0.0.0.0:1100
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: tcp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: tcp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: UDP relay enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: UDP relay enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: udp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: udp port reuse enabled
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6322]: running from root user
Sun Jul 22 20:33:44 2018 daemon.info /usr/bin/ss-redir[6323]: running from root user
Sun Jul 22 20:33:44 2018 user.notice mptcp: Reloading mptcp config due to ifup of omr6in4 (6in4-omr6in4)
Sun Jul 22 20:33:45 2018 user.notice firewall: Reloading firewall due to ifup of WAN_LTE (wwan0)
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-local[6708]: using tcp fast open
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-local[6708]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:46 2018 user.info : using tcp fast open
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-redir[6709]: resolving hostname to IPv6 address first
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-redir[6709]: initializing ciphers... aes-256-cfb
Sun Jul 22 20:33:46 2018 user.info : using tcp fast open
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-redir[6710]: resolving hostname to IPv6 address first
Sun Jul 22 20:33:46 2018 daemon.info /usr/bin/ss-redir[6710]: initializing ciphers... aes-256-cfb
[... cut ...]
Sun Jul 22 20:36:09 2018 daemon.err /usr/bin/ss-local[7584]: getpeername: Socket not connected
Sun Jul 22 20:36:30 2018 daemon.err /usr/bin/ss-local[7584]: getpeername: Socket not connected
Sun Jul 22 20:36:50 2018 daemon.err /usr/bin/ss-local[7584]: getpeername: Socket not connected
Sun Jul 22 20:36:52 2018 daemon.err uhttpd[3545]: luci: accepted login on /admin/network/network for root from 192.168.100.2
Sun Jul 22 20:37:11 2018 daemon.err /usr/bin/ss-local[7584]: getpeername: Socket not connected

Ysurac, 2018-07-22:
I don't know if this is the only problem but you have a power problem (*Voltage normalised (0x00000000)*)

lelectronlibre, 2018-07-22:
Strange because everything work before tun0 interface turns on. I will try with powered USB hub, maybe it will solve the problem.

Ysurac, 2018-07-22:
You can try to disable glorytun also.

lelectronlibre, 2018-07-22:
I try powered USB hub, no more voltage error but nothing new for internet access.

I disabled glorytun and re-enabled it and there is a daemon error : 

Sun Jul 22 21:28:38 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Jul 22 21:28:38 2018 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jul 22 21:28:38 2018 daemon.notice netifd: Network device 'tun0' link is up
Sun Jul 22 21:28:38 2018 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Jul 22 21:28:38 2018 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Jul 22 21:28:38 2018 daemon.info glorytun[11755]: INITIALIZED tun0
Sun Jul 22 21:28:38 2018 daemon.notice netifd: omrvpn (11792): udhcpc: started, v1.28.4
Sun Jul 22 21:28:38 2018 daemon.notice netifd: omrvpn (11792): udhcpc: sending discover
Sun Jul 22 21:28:39 2018 **daemon.err glorytun[11755]: MyIPServer.65001: connected**
Sun Jul 22 21:28:39 2018 daemon.info glorytun[11755]: STARTED tun0
Sun Jul 22 21:28:40 2018 daemon.notice netifd: omrvpn (11792): udhcpc: sending select for 10.255.255.2
Sun Jul 22 21:28:40 2018 daemon.notice netifd: omrvpn (11792): udhcpc: lease of 10.255.255.2 obtained, lease time 43200
Sun Jul 22 21:28:41 2018 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Jul 22 21:28:41 2018 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: using tcp fast open
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: initializing ciphers... aes-256-cfb
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: listening at 127.0.0.1:1111
Sun Jul 22 21:28:43 2018 user.info : using tcp fast open
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: tcp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: resolving hostname to IPv6 address first
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: udprelay enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: initializing ciphers... aes-256-cfb
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: udp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-local[12338]: running from root user
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: listening at :::1100
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: tcp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: UDP relay enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: udp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12339]: running from root user
Sun Jul 22 21:28:43 2018 user.info : using tcp fast open
Sun Jul 22 21:28:43 2018 user.info : using tcp fast open
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: resolving hostname to IPv6 address first
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: initializing ciphers... aes-256-cfb
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: listening at 0.0.0.0:1100
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: tcp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: UDP relay enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: udp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12340]: running from root user
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: initializing ciphers... aes-256-cfb
Sun Jul 22 21:28:43 2018 user.info : using tcp fast open
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: initializing ciphers... aes-256-cfb
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: listening at 0.0.0.0:1100
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: tcp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: UDP relay enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: udp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12342]: running from root user
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: listening at 0.0.0.0:1100
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: tcp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: UDP relay enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: udp port reuse enabled
Sun Jul 22 21:28:43 2018 daemon.info /usr/bin/ss-redir[12341]: running from root user
Sun Jul 22 21:28:43 2018 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jul 22 21:29:27 2018 daemon.err glorytun[11755]: write: Operation timed out
Sun Jul 22 21:29:27 2018 daemon.info glorytun[11755]: STOPPED tun0
Sun Jul 22 21:29:58 2018 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 104.40.211.35)

Ysurac, 2018-07-22:
Disable multipath on wan1 and wan2 (or at least make sure only one interface is set as master).
You can also check configuration on System->OpenMPTCProuter->Status page.

lelectronlibre, 2018-07-22:
Disable multipath has no effect. In network file, can I have to modify config globals also : 
config globals 'globals'
option ula_prefix 'fd6f:dd65:b46e::/48'
option multipath 'enable'
option mptcp_path_manager 'fullmesh'
option mptcp_scheduler 'default'
option congestion 'olia'
option mptcp_checksum '1'
option mptcp_syn_retries '30'
option mptcp_fullmesh_num_subflows '1'
option mptcp_fullmesh_create_on_err '1'
option mptcp_ndiffports_num_subflows '1'

My actual OpenMPTPCrouter status page with internet access working (without glorytun) :  
[https://i.imgur.com/zwrjb5r.png](url)

Ysurac, 2018-07-23:
I have no success to detect IP and gateway. Can you give me the result of ```ip route``` and ```ip addr``` on OpenMPTCProuter ?

Ysurac, 2018-07-23:
Could you give me the result of ```ubus call network.interface dump``` ? I think the problem is in this result.

lelectronlibre, 2018-07-23:
Of course, result : 

{
	"interface": [
		{
			"interface": "WAN_LTE",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"dynamic": false,
			"uptime": 3980,
			"l3_device": "wwan0",
			"proto": "ncm",
			"updated": [
				"data"
			],
			"metric": 0,
			"dns_metric": 0,
			"delegation": true,
			"ipv4-address": [
				
			],
			"ipv6-address": [
				
			],
			"ipv6-prefix": [
				
			],
			"ipv6-prefix-assignment": [
				
			],
			"route": [
				
			],
			"dns-server": [
				
			],
			"dns-search": [
				
			],
			"inactive": {
				"ipv4-address": [
					
				],
				"ipv6-address": [
					
				],
				"route": [
					
				],
				"dns-server": [
					
				],
				"dns-search": [
					
				]
			},
			"data": {
				"manufacturer": "huawei"
			}
		},
		{
			"interface": "WAN_LTE_4",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"dynamic": true,
			"uptime": 3977,
[... 375 lines cut ...]
				
			],
			"inactive": {
				"ipv4-address": [
					
				],
				"ipv6-address": [
					
				],
				"route": [
					
				],
				"dns-server": [
					
				],
				"dns-search": [
					
				]
			},
			"data": {
				
			}
		}
	]
}

Ysurac, 2018-07-23:
Strange WAN_LTE is empty, data are in WAN_LTE_4. But my scripts only look for WAN_LTE.
What you get with ```ifstatus WAN_LTE``` same empty data ?

Ysurac, 2018-07-23:
ok thanks.
The interface with "_4" is created by ncm. This will be supported in next version (that should be out soon).

lelectronlibre, 2018-07-23:
Ok great !!
Thank you for your help and your beautiful work on OpenMPTCProuter !!!

PS : Sorry for my poor english during our session, I'm French... ;)

Ysurac, 2018-07-30:
Should work in latest version.

closed 2018-07-30 by Ysurac