Ysurac/openmptcprouter #109: Clients can't do ICMP nor 3rd parties DNS

PxxBxx, 2018-07-28:
## Expected Behavior

#### Clients on the LAN should be able to ping IPs outside the setup.
 eg: a local host with cacti/munin/nagios/... applications has to do ping request and get echo

#### Clients on the LAN should be able to access any DNS server
 eg: a Chromecast device has the Google DNS hardcoded in their software

## Actual Behavior

#### Ping (ICMP) is not working inside the lan area.
eg: my laptop (ip 192.168.100.128 gw 192.168.100.1) can't do any ping
```
PING 8.8.8.8 (8.8.8.8): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
```
On the VPS Side here are some glorytun logs (syslog)
_Note : tcpdump does not show anything when filtering on 8.8.8.8_
* Glorytun TCP
```
glorytun-tcp-run[506]: xx.xx.xx.xx.32632: connected
kernel: [ 4562.350656] IPv4: martian source 8.8.8.8 from 192.168.100.128, on dev gt-tun0
```
* Glorytun UDP
```
kernel: [ 4501.297061] IPv4: martian source 8.8.8.8 from 192.168.100.128, on dev gt-udp-tun0
```

#### DNS Access seems to be blocked someway
```
dig @8.8.8.8 google.com

; <<>> DiG 9.8.3-P1 <<>> @8.8.8.8 google.com
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached
```
(no logs available

## Steps to Reproduce the Problem

  1. Install a default v0.37 VPS, set the keys (ss and gt), reboot
  2. Install a default v0.29 RPI3, set the keys (ss and gt)
  3. Set the WAN addresses on OMR
  4. connect a client to the router and do ping and dig requests

## Specifications

  - OpenMPTCProuter version: v0.29
  - OpenMPTCProuter VPS version: v0.37
  - OpenMPTCProuter platform: RPI3

PxxBxx, 2018-07-28:
The OMR status page seems good, it only tells "no IPv6 access" because I tried to disable IPv6 beforehand, with no more success.
The behaviour is the same if I ping any other IPs (eg. other VPS I own), or try to dig DNS other than google (eg. my ISPs).
Maybe the package with shorewall rules does not match your own ? I tried an upgrade on the VPS, but it didn't change anything (no ping, no dig DNS)

Ysurac, 2018-07-28:
I've updated OMR VPS script. This may solve your issue.

PxxBxx, 2018-07-28:
Hello, I tried the new VPS script on a fresh debian9, the script doesn't stop anymore around OMR service, but no enhancement to my issue (no ping, no external DNS).
Furthermore, I detected I could not even SSH to my VPS from a OMR client (I always used my local server to connect (port 65222), which is not connected on OMR, but directly on one of the wan).

I'm beginning to hate firewalls, I'm pretty sure it's related...
Could you tell me how to deactivate shorewall on the VPS, but keeping only the relevant functionnal rules for ShadowSocks/Glorytun/snat/... ? (to test if it's related, at least on the VPS side, I don't care about firewall/security/...)

Ysurac, 2018-07-28:
All rules are relevant :)

Could you give me your ```/etc/config/network``` file ?

PxxBxx, 2018-07-28:
of course ! (I deleted wan3 on eth1, I didn't want usb-modeswitch 4G/router to interfere, but it didn't change anything)

```
root@OpenMPTCProuter:~# cat /etc/config/network

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option multipath 'off'

config globals 'globals'
	option ula_prefix 'fd36:cb99:c67e::/48'
	option multipath 'enable'
	option mptcp_path_manager 'fullmesh'
	option mptcp_scheduler 'default'
	option congestion 'olia'
	option mptcp_checksum '0'
	option mptcp_syn_retries '5'
	option mptcp_fullmesh_num_subflows '1'
	option mptcp_fullmesh_create_on_err '1'
	option mptcp_ndiffports_num_subflows '1'

config interface 'lan'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option metric '2048'
	option multipath 'off'
	option ip4table 'lan'

config rule 'lan_rule'
	option lookup 'lan'
	option priority '100'

config interface 'wan1'
	option proto 'static'
	option ip4table 'wan'
	option multipath 'master'
	option defaultroute '0'
	option type 'macvlan'
	option ifname 'wan1'
	option ipaddr '192.168.1.222'
	option netmask '255.255.255.0'
	option gateway '192.168.1.1'

config device 'wan1_dev'
	option name 'wan1'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1532516643'

config route6
	option interface 'lan'
	option target '::/0'

config interface 'omrvpn'
	option ifname 'tun0'
	option proto 'dhcp'
	option ip4table 'vpn'
	option multipath 'off'
	option leasetime '12h'
	option mtu '1400'

config interface 'omr6in4'
	option proto '6in4'
	option ip4table 'vpn'
	option multipath 'off'
	option ipaddr '10.255.255.2'
	option peeraddr '10.255.255.1'
	option gateway 'fe80::a00:1'
	option ip6addr 'fe80::a00:2'

config interface 'wan2'
	option proto 'static'
	option type 'macvlan'
	option ip4table 'wan'
	option multipath 'on'
	option defaultroute '0'
	option ifname 'wan2'
	option ipaddr '192.168.3.222'
	option netmask '255.255.255.0'
	option gateway '192.168.3.1'

config device 'wan2_dev'
	option name 'wan2'
	option type 'macvlan'
	option ifname 'eth0'
	option macaddr 'auto1532724593'
```

Ysurac, 2018-07-28:
ok no problem here.
Can I get ```/etc/config/firewall``` ?

SoGooDFR, 2018-07-28:
I confirm the bug. GlorytunTCP/UDP is UP, the tunnel is ok but on teamspeak : server refused the connection. Same problem when i try to access to the vps from VPN tunnel.

Ysurac, 2018-07-28:
On the VPS side, I would need to know if omr service is running: ```ps aux | grep omr```. If yes I would also need ```/etc/shorewall/params.vpn``` on the VPS.

PxxBxx, 2018-07-29:
Hello,
## UPDATE, I'VE JUST INSTALLED v0.29.1 for RPI3
openmptcprouter-v0.29.1-r7171-962e86d9af-brcm2708-bcm2710-rpi-3-squashfs-sdcard.img

  * clean install
no dhcp detected by clients :'( (I had this on v0.29)
```
$ sudo nmap --script broadcast-dhcp-discover -e eth0

Starting Nmap 7.60 ( https://nmap.org ) at 2018-07-29 10:28 CEST
WARNING: No targets were specified, so 0 hosts scanned.
Nmap done: 0 IP addresses (0 hosts up) scanned in 10.49 seconds
```
  * manual IP, connect to OMR, setup VPS IP, SS&GT keys, wan1
TCP OK, but no ping nor icmp (nor ssh to vps)
  * setup wan2
same, TCP OK, but no ping nor icmp (nor ssh to vps)
  * reboot OMR
still no DHCP from clients
  * untick DHCP authoritative, save, re-tick DHCP authoritative
got DHCP, and **ping and dig and ssh to vps**
  * reboot OMR
not better, no dhcp
  * take a coffee and chill (let's say 5-10min)
**everything works, dhcp, SS+GT-TCP, ping, dig, ssh to vps, chromecast, switch, ...**
```
$ sudo nmap --script broadcast-dhcp-discover -e eth0

Starting Nmap 7.60 ( https://nmap.org ) at 2018-07-29 11:00 CEST
Pre-scan script results:
| broadcast-dhcp-discover:
|   Response 1 of 1:
|     IP Offered: 192.168.100.249
|     DHCP Message Type: DHCPOFFER
|     Server Identifier: 192.168.100.1
|     IP Address Lease Time: 2m00s
|     Renewal Time Value: 1m00s
|     Rebinding Time Value: 1m45s
|     Subnet Mask: 255.255.255.0
|     Broadcast Address: 192.168.100.255
|     Router: 192.168.100.1
|     Domain Name Server: 192.168.100.1
|_    Domain Name: lan
```

  * plug usb 4G router (modeswitch), setup wan3, modify /etc/config/network to set eth1 for wan3, then reboot and wait 5min
Still everything OK

**### If I stop there, the issue is closed after v0.29.1.**
_But everything I did (even wait for coffee and chill) was also done with v0.29, with no success._
So I think that what you @Ysurac did with dnsmasq init had something to do (in the positive way)
Having no DHCP in the first few minutes is strange. not problematic, but strange.

I'll backup my config and will start a new install on the RPI, to see if I get more luck.
But I'm so glad to see everything working, thanks for the v0.29.1 update !!

closed 2018-08-02 by Ysurac